micro-manager/micro-manager #39: Viewer gets confused after using a multi-camera device

nicost, 2015-07-07:
Use a multi-camera configuration.  Snap an image with the multi-camera device (that has two or more channels).  Viewer shows the expected number of channels.  Switch to a single camera.  Snap an image.  Viewer still shows the two or more channels.  Close the Snap/Live View window.  Snap an image.  Viewer still shows the two or more channels, not the single channel coming from our camera

ChrisWeisiger, 2015-07-07:
Just to be clear, this is on the mm2 branch?

-Chris

nicost, 2015-07-07:
Absolutely!

Does one need to indicate that somehow?  Isn't this repository all about 
the mm2 branch?

Best,

Nico

ChrisWeisiger, 2015-07-07:
Technically the repository has a copy of the master branch, even if it's
not necessarily up-to-date. And people could easily report 1.4 bugs here
instead of through other avenues, so I figure it's best to be explicit.

I'll take a look at this bug tomorrow. Sounds like the Snap/Live Manager
isn't properly recognizing when it needs to throw away its old datastore
and create a new one.

-Chris

omsai, 2015-07-08:
>> Isn't this repository all about the mm2 branch?
>
> I figure it's best to be explicit.

Raised #40.

ChrisWeisiger, 2015-07-08:
Fixed in commit e3cad32c.

closed 2015-07-08 by ChrisWeisiger