micro-manager/micro-manager #40: [Meta] Issue reporting guidelines for community submissions

omsai, 2015-07-08:
As GitHub allows anyone to report an issue, unlike Trac, it's probably best to have a document with guidelines we can point to about reporting issues.

Not sure where the best place to put that document would be: in the repo, MediaWiki, or if you want to go full meta - an issue on how to report issues!  Ideally the guidelines / structure would be visible when creating an issue from GitHub, but this doesn't seem possible from a cursory web search and read of the GitHub issue API.

While labels are okay for some degree of organization, the title might include the DeviceAdapter name (so we don't have 160+ labels).  Maybe the title could also specify the MM version instead of the description, if searching by title allows globbing.

ChrisWeisiger, 2015-07-08:
Let's not go adding formality where none may be needed. If we observe
people consistently having trouble reporting issues in a useful manner,
then we can see about setting up guidelines and figuring out how to enforce
them.

omsai, 2015-11-21:
Sounds good!

closed 2015-11-21 by omsai