micro-manager/micro-manager #42: top window does not update

samjlord, 2015-07-09:
When I adjust the histogram display range, the top window does not update unless I click onto a different window and then click back. When I do click back, the display does update.

This seems to only be an issue for an "Album" window, not with a Live window.

ChrisWeisiger, 2015-07-09:
Hm, I can't reproduce this on my system. Are you in grayscale, color,
composite view? If multi-channel, how many channels? What dimensions are
the image? I'm guessing you're on Windows?

Thanks.

samjlord, 2015-07-09:
I'm using grayscale, single channel, single dimension. Yes, on Windows 7.
Feel free to come downstairs to try it on our system here sometime.

-Sam

ChrisWeisiger, 2015-07-16:
Hi Sam,

Just wanted to let you know I haven't forgotten about this. I've just been
wrestling with other high-priority issues. But I will get back to you,
hopefully early next week.

-Chris

ChrisWeisiger, 2015-07-16:
Actually, could you try downloading tonight's nightly build and seeing if
that fixes it? I haven't *specifically* changed anything related to
grayscale images, but the display routines have seen significant work in
the past week, so it's possible that whatever you're facing has been fixed
as a side benefit. Thanks!

-Chris

closed 2015-08-05 by ChrisWeisiger