micro-manager/micro-manager #48: mm2: Linking display settings for MDA coordinates

marktsuchida, 2015-08-19:
Run MDA with 10 time points, 2 channels (call this "(a)").
In inspector, set Color mode to Color.
Select Gear > Duplicate This Window (call this "(b)").
At this point, image in (a) is temporarily displayed in grayscale (bug).
In window (b), use link button to link time point to window (a).
This causes window (a) to jump to the same time point as (b) (correct) but also causes the channel in (a) to jump to the channel displayed in (b) (spurious).
Now use scroll bar to show channel 1 in (a) and channel 2 in (b).
Use Play button in (b) to animate time points (works correctly). Stop.
Use Play button in (a) to animate time points. Channel in (b) jumps to channel 1 (bug).
Not sure if this is specific to channel axis or other axes are also affected.

marktsuchida, 2015-08-19:
After the above, close window (b). Window (a) is again temporarily displayed in grayscale, until it is clicked, at which point it fixes itself.

marktsuchida, 2015-08-19:
The temporary grayscale rendering appears to be a distinct issue; opened #49. Let's limit this issue to the coordinate linking.

ChrisWeisiger, 2015-09-15:
This is fixed in 85baa927e6c6d3153077f7ab673fcf7361583f52.

closed 2015-09-15 by ChrisWeisiger