micro-manager/micro-manager #49: mm2: Image temporarily drawn in grayscale after certain operations

marktsuchida, 2015-08-19:
When a multi-channel image is set to Color mode = Color, the image gets incorrectly rendered in grayscale when

- the zoom buttons are clicked, or
- the window is duplicated (Gear > Duplicate This Window)

until the window is clicked again. Seen on OS X.

ChrisWeisiger, 2015-09-11:
This is crudely fixed now; after those specific operations occur we manually re-apply our LUTs. I have no idea what was causing the original issue; ideally we'd fix that, but there's decent odds it's in the depths of ImageJ, and investigations can take a long time. So I'm just shelving this with the workaround. If we ever want to fix the underlying cause, there's a "HACK" tag in the code to mark it.

ChrisWeisiger, 2015-09-11:
Not actually fixed, alas; we now switch to Color mode when duplicating displays, until the histograms are updated. What the heck. Incorrect drawing after zoom appears to be fixed, at least.

ChrisWeisiger, 2015-09-11:
This may potentially be related to multicamera setups; when I run an MDA with a single camera and multiple channels, the colors behave themselves more often. Except:

1) Run MDA with two channels, ten timepoints. Set display mode to color.
2) Duplicate display, play with linking time axis.
3) Close duplicated display.
4) Original display switches to grayscale.

ChrisWeisiger, 2015-09-15:
Another repro case:

1) Run MDA with two channels, ten timepoints. Set display mode to color.
2) Close inspector window.
3) Duplicate display. Duplicated display is in grayscale.

ChrisWeisiger, 2015-09-17:
Okay, I think this is finally fixed. But given the nature of the "fix" (manually reapplying LUTs in situations where we spontaneously shift to grayscale), it's entirely possible that there are edge cases where we still lose our colors. Fixing that would require knowing why we end up in grayscale in the first place. Given that we aren't explicitly telling the display to go gray, I have to assume this is some kind of ImageJ bug, and I don't think we have the resources to investigate that right now.

closed 2015-09-17 by ChrisWeisiger