micro-manager/micro-manager #52: mm2: Image viewer color incorrectly initialized?

marktsuchida, 2015-08-19:
Start with fresh user profile and standard demo config.

MDA, 6 time points, Z stack of 0-5 at step = 1, channel group "Channel", create two channels. Run MDA.

The assigned channels colors are blue and orange, and the inspector says Color mode is Grayscale, but the actual image is rendered in colors that do not match either.

Tested on OS X.

ChrisWeisiger, 2015-09-10:
Improved in 1e23ac259c2588cd4d7448bb8dffc5d9bf1df01f. Now the channel colors are only briefly wrong and are fixed as soon as we do an update of the display. Hopefully we will eventually get a proper fix so the display is never wrong.

ChrisWeisiger, 2015-09-15:
It appears that commit 1ba5576f1279418d5297a008d2abca57ae8b87ef has fully fixed this.

closed 2015-09-15 by ChrisWeisiger