micro-manager/micro-manager #62: Size of main window is not remembered through re-start

nicost, 2015-09-22:
Start MM.  Change the size of the main window (for instance, to show all the Configuration groups).  Close the window (quit MM). Re-start MM.  Note that the main window opens in its original size, not the desired re-sized size.

ChrisWeisiger, 2015-09-22:
Is it specifically that the window size should be remembered, or should we instead try to ensure that the window is automatically large enough to hold all of the configuration groups (and the bits of the window below the config group pane)?

nicost, 2015-09-22:
If we let users resize the window, then the window should open again in the same size upon restart.  I do not think that should take away the ability to resize the window (someone may want it to be small), hence I think that only the size should be remembered and not attempts should be made to try to second guess the user.

closed 2015-09-22 by ChrisWeisiger