micro-manager/micro-manager #67: Projector plugin: Calibration is not stored, hence the plugin can not be used.

nicost, 2015-09-24:
Open the Projector plugin.  Go to Setup tab.  Press Calibrate! button.  Wait for calibration to succeed (you need hardware for this).  Click on Point and Shoot and follow the instructions.  The plugin shows an error message with the text: "Error: Please calibrate the phototargeting device first, using theSetup tab."

ChrisWeisiger, 2015-09-28:
To be clear, the Projector plugin shows you a dialog saying "Calibration finished" (and not "Calibration canceled"), correct? Could you send me a problem report in which you run calibration? It's possible an error occurred either during saving of the set of transforms or during later loading of those transforms. Otherwise, I don't see anything obviously wrong with the code, so fixing this is going to require debugging on hardware.

nicost, 2015-09-28:
On 9/28/15 10:43 AM, ChrisWeisiger wrote:
>
> To be clear, the Projector plugin shows you a dialog saying 
> "Calibration finished" (and not "Calibration canceled"), correct?
>
Correct.

> Could you send me a problem report in which you run calibration? It's
> possible an error occurred either during saving of the set of
> transforms or during later loading of those transforms. [...]
You can replicate this using the Demo Galvo device.  Add it to your 
configuration.  Make sure to set the DemoCamera Mode property to 
"Noise".  You can then run the Projector plugin and Calibrate.  I ended 
up setting the Camera exposure time to 10 ms, the Projector Exposure 
time to 1 ms, and the Projector Delay to 10 ms. Calibration ends with 
"Calibration Finished", but no error. There are lots of debug messages 
about failure to extract metadata. Point and Shoot fails with the 
"Please calibrate the phototargeting device first " error message.

Seems that the error can be nicely re-produced without hardware.

Best,

Nico

ChrisWeisiger, 2015-09-30:
This has been fixed in 70c803e3484e97aa06074178cb295df63d79cc8c.

closed 2015-09-30 by ChrisWeisiger