micro-manager/micro-manager #78: mm2: Memory leak in live mode

nicost, 2015-10-06:
When running live mode with an Andor Zyla, memory usage (as judged by the Windows Task Manager. goes up from about 200MB t0 > 4GB, then drops and grows again.  Memory usage as reported by Java is much lower, but has a similar sharp drop when the Windows tool shows the drop, suggesting that this is memory in direct byte buffers that is being kept by data structured in the Java layer that are garbage collected.  This is a problem on systems that do not have enough RAM, such as 32-bit Windows.  On 32-bit Windows I see live mode stopping (without an error message) and the logs showing that it failed to acquire memory for direct byte buffers.  This does not happen on Micro-Manager 1.4.

ChrisWeisiger, 2015-10-06:
Okay, what's most likely happening here is that each new image we create is allocating a new direct buffer, getting stored (briefly) in the Live mode's Datastore (which only ever stores one image per channel), and then immediately getting overwritten by the next incoming image. Memory usage drops when the garbage collector runs, but that's evidently not happening frequently enough for 32-bit systems.

So our options, as I see it, are either to manually run the GC periodically, or figure out some way to re-use the direct buffers so we aren't constantly reallocating more. The first is obviously not ideal, but the second may be tricky to implement.

I would expect that saving to disk during an MDA would also have this problem, as images are briefly created for display and then forgotten about when the next image comes along.

The lack of error message is a related issue that also needs to be solved.

Thanks for the report, Nico!

ChrisWeisiger, 2015-10-08:
This should be ameliorated by commit 463855e7b6b880fdaedd6cc47b2920bc7e7a7394, which limits the rate at which Live mode attempts to pull images out of the Core, based on the observed rate at which images are able to be displayed on the screen. Previously, live mode had a lot of "churn" when the camera was fast but the display was slow; this churn caused it to create and immediately discard large numbers of images (and thus of native byte buffers).

Note that this amelioration is currently only enabled if you have no data processing pipeline set up. Still, it should help.

closed 2015-10-08 by ChrisWeisiger

nicost, 2015-10-11:
I can confirm that live mode now works on Windows 7 32-bit using an 
Andor Zyla camera.  Thanks!