micro-manager/micro-manager #82: Per-camera (or per-channel) settings for Image Flipper

marktsuchida, 2015-10-13:
See http://micro-manager.3463995.n2.nabble.com/Mirroring-Transposing-Image-in-Live-View-tp5951388p7585778.html

This is surely going to come up for other processors. It might make sense to create a general way to apply a processor only to a certain channel. This can either be built into the pipeline system, or maybe there could be a meta-processor that wraps the raw processor.

ChrisWeisiger, 2015-10-16:
The Image Flipper already allows you to select which camera to apply it to. Do we expect many users to have setups where the channel selection transposes/flips the image?

Maybe the right thing to do here is have a Filter meta-processor that only runs the "contained" processor if specific attributes of the image metadata have specific values (or perhaps if a given configuration group is set to a specified mode, though I don't think that's transparently shown in the image metadata). Making that GUI easy to use would be very tricky. It would provide the maximum flexibility though. It would also require images to have the full scope data even when run outside of acquisitions, though (which is not currently the case).

marktsuchida, 2015-10-16:
This is a case of me looking back at an old email and misinterpreting. Never mind...
(What was requested was a faster flipper implementation, so that will need to be done in the Core.)

closed 2015-10-16 by marktsuchida