micro-manager/micro-manager #83: Quick Access Window features

ChrisWeisiger, 2015-10-26:
Feedback from Dane:

 * Be able to select a custom icon or at least background color for the Presets button
 * Make buttons bigger
 * Add a button to load a specific set of acquisition settings and then run that acquisition
 * (possible) be able to have multiple of these windows so you can e.g. have button bars on the left and bottom edges of the screen.

ChrisWeisiger, 2015-10-29:
Aaron suggests being able to add column/row headers, separators, etc. so the window can be visually grouped in configurable ways. He'd also like the ability to have multiple of these windows.

ChrisWeisiger, 2015-11-10:
Presets now have background colors which, when possible, automatically pull from the channel colors. Buttons are bigger. There's a "run saved acquisition" button, and a free-text field. That's the easy enhancements done.

ChrisWeisiger, 2015-11-10:
Aaron would like to be able to rename the windows themselves if multiple windows are allowed. Also, a widget that lets you select experiment settings from a dropdown menu (and ditto for scripts), with a run button. Possibly do these as adaptations of the existing "run saved acquisition" and "run script" widgets, depending on if the user selects a file or a folder? That might be too obtuse.

ChrisWeisiger, 2015-11-11:
The ability to assign custom icons to each button would be helpful. Ideally this would be any button, though I'm not certain how exactly that would work as most of those implement the WidgetPlugin interface, which is super-generic. Maybe we can add an "assignIcon" method that the WidgetPlugin is free to do whatever with? We would need a way to give the plugin access to the controls it has created (since it should not itself keep track of those controls), though...

ChrisWeisiger, 2015-11-12:
Multiple windows are now supported.

ChrisWeisiger, 2015-11-16:
Customizable icons are now supported. Users can select from a variety of built-in icons, or a color swatch, or a loaded image from a file.

ChrisWeisiger, 2015-11-16:
Titles can now be customized. I believe that's everything from the current wishlist.

ChrisWeisiger, 2015-11-19:
New requests: 

 * widget that shows the current state of a specific hardware property (and/or allows modifying said state)
 * Add "close all windows" button widget (and add to main window)
 * When a Quick Access Panel window is closed, it should leave configuration mode (so it's not in configuration mode when re-opened).
 * Add some guidance to configuration mode to tell users how to return to normal operation.
 * Add widget to switch to a preset and open the shutter?

ChrisWeisiger, 2015-11-19:
Added "close all windows" widget. Closed Quick Access Panels leave configure mode.

marktsuchida, 2020-08-28:
Many of the above have been implemented. New issues should be filed for any remaining that are desired. (Although I feel that some global design thinking (including division of roles with the main window) is more called for than any individual feature.)

closed 2020-08-28 by marktsuchida