micro-manager/micro-manager #87: Changes in gamma in the histogram are no longer propagated to the image.

nicost, 2015-10-30:
Snap an image.  In the histogram in the Image Inspector, grab the line showing the display function and drag it up or down to change the gamma.  Note that there is no change in the displayed image.

ChrisWeisiger, 2015-11-04:
I am not able to replicate this, so I assume it was fixed as a side-effect of one of the many histogram changes made in the last week.

nicost, 2015-11-04:
Still the same problem for me.  Gamma seems to work in "Color" mode, but 
not with any of the other lookup tables (including "Grayscale");

Also, it is easy to drag the gamma slider all the way to the top by 
accident.  If you do so, there is no way to get it back.

Open Recent menu entry no longer shows the full path (which was a really 
nice way to remember where those files were stored and will also help 
when opening datasets with the same name in different locations).

Opening a dataset through the "Open Recent (RAM) " menu hangs MM 
(whereas opening through "Open (Ram)" is fine.

In a 3 color dataset in composite mode, the histogram sliders do not 
work as expected:
- slider channel 1: does nothing
- slider channel 2: works as expected
- slider channel 3: operates on channel 1

I put the dataset tested at: 
http://valelab.ucsf.edu/~nstuurman/images/BPAECells2/

In composite mode, the "eye" icons do not work (after closing the eye, 
the corresponding channel is still visible.

When closing the Inspector window and re-opening it, I get:

2015-11-04T10:07:15.083427 tid0x127bc8000 [IFO,App] [UI] MMImageCanvas 
"canvas0" clicked in DefaultDisplayWindow.
2015-11-04T10:07:15.702935 tid0x127bc8000 [IFO,App] [UI] GearButton 
clicked in DefaultDisplayWindow.
2015-11-04T10:07:16.924241 tid0x127bc8000 [IFO,App] Thread 14 
(AWT-EventQueue-0) terminated with uncaught exception
2015-11-04T10:07:16.924268 tid0x127bc8000 [IFO,App] 
java.lang.NoSuchMethodError: 
org.micromanager.display.internal.inspector.InspectorFrame.<init>(Lorg/micromanager/display/DisplayWindow;)V
2015-11-04T10:07:16.924352 tid0x127bc8000 [IFO,App]   at 
org.micromanager.display.internal.GearButton$1.actionPerformed(GearButton.java:52)
2015-11-04T10:07:16.924358 tid0x127bc8000 [IFO,App]   at 
javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
2015-11-04T10:07:16.924362 tid0x127bc8000 [IFO,App]   at 
javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2346)
2015-11-04T10:07:16.924366 tid0x127bc8000 [IFO,App]   at 
javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
2015-11-04T10:07:16.924370 tid0x127bc8000 [IFO,App]   at 
javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
2015-11-04T10:07:16.924373 tid0x127bc8000 [IFO,App]   at 
javax.swing.AbstractButton.doClick(AbstractButton.java:376)
2015-11-04T10:07:16.924377 tid0x127bc8000 [IFO,App]   at 
javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:833)
2015-11-04T10:07:16.924381 tid0x127bc8000 [IFO,App]   at 
com.apple.laf.AquaMenuItemUI.doClick(AquaMenuItemUI.java:157)
2015-11-04T10:07:16.924385 tid0x127bc8000 [IFO,App]   at 
javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:877)
2015-11-04T10:07:16.924389 tid0x127bc8000 [IFO,App]   at 
java.awt.Component.processMouseEvent(Component.java:6525)
2015-11-04T10:07:16.924392 tid0x127bc8000 [IFO,App]   at 
javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
2015-11-04T10:07:16.924396 tid0x127bc8000 [IFO,App]   at 
java.awt.Component.processEvent(Component.java:6290)
2015-11-04T10:07:16.924399 tid0x127bc8000 [IFO,App]   at 
java.awt.Container.processEvent(Container.java:2234)
2015-11-04T10:07:16.924403 tid0x127bc8000 [IFO,App]   at 
java.awt.Component.dispatchEventImpl(Component.java:4881)
2015-11-04T10:07:16.924406 tid0x127bc8000 [IFO,App]   at 
java.awt.Container.dispatchEventImpl(Container.java:2292)
2015-11-04T10:07:16.924498 tid0x127bc8000 [IFO,App]   at 
java.awt.Component.dispatchEvent(Component.java:4703)
2015-11-04T10:07:16.924504 tid0x127bc8000 [IFO,App]   at 
java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4898)
[... 21 lines cut ...]
java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
2015-11-04T10:07:16.924541 tid0x127bc8000 [IFO,App]   at 
java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:86)
2015-11-04T10:07:16.924543 tid0x127bc8000 [IFO,App]   at 
java.awt.EventQueue$4.run(EventQueue.java:731)
2015-11-04T10:07:16.924545 tid0x127bc8000 [IFO,App]   at 
java.awt.EventQueue$4.run(EventQueue.java:729)
2015-11-04T10:07:16.924566 tid0x127bc8000 [IFO,App]   at 
java.security.AccessController.doPrivileged(Native Method)
2015-11-04T10:07:16.924571 tid0x127bc8000 [IFO,App]   at 
java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
2015-11-04T10:07:16.924574 tid0x127bc8000 [IFO,App]   at 
java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
2015-11-04T10:07:16.924576 tid0x127bc8000 [IFO,App]   at 
java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
2015-11-04T10:07:16.924578 tid0x127bc8000 [IFO,App]   at 
java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
2015-11-04T10:07:16.924581 tid0x127bc8000 [IFO,App]   at 
java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
2015-11-04T10:07:16.924583 tid0x127bc8000 [IFO,App]   at 
java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
2015-11-04T10:07:16.924586 tid0x127bc8000 [IFO,App]   at 
java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
2015-11-04T10:07:16.924588 tid0x127bc8000 [IFO,App]   at 
java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

ChrisWeisiger, 2015-11-04:
Wow, okay. I'll definitely take a look. Thanks.

Were you running the nightly build generated last night?

nicost, 2015-11-04:
On 11/4/15 10:28 AM, ChrisWeisiger wrote:
>
> Wow, okay. I'll definitely take a look. Thanks.
>
> Were you running the nightly build generated last night?
>
Fresh compile after commit applying Flak Detmar's patch.

Best,

Nico

ChrisWeisiger, 2015-11-04:
Good to know, thanks.

I've fixed the gamma curve not working in grayscale and the gamma curve becoming so extreme you can't adjust it any more (it now snaps back to a straight curve if you drag it too far). The others will take a bit longer.

ChrisWeisiger, 2015-11-04:
EDT hang from Open Recent menu is fixed.

ChrisWeisiger, 2015-11-04:
Eye buttons have been fixed. I'm unable to replicate the Inspector window re-open bug. My guess is that you need to do a clean build, though; it sounds like one of the classes changed its allowed constructors and another class hasn't picked up the change yet.

I've noticed another bug: when opening the file you provided, channel colors aren't loaded properly. Histograms are drawn with white colors and the image display defaults to the colorblind-friendly colors. Arguably this isn't *completely* a bug since you didn't include your display settings file with the TIFF (which would normally be used to reconstruct our display windows when loading the saved data), but we should try to fall back to the data in the multipage TIFF, and if nothing else the histograms should not draw as white.

nicost, 2015-11-05:
> Eye buttons have been fixed.
>
Confirmed.

> I'm unable to replicate the Inspector window re-open bug. My guess is 
> that you need to do a clean build, though; it sounds like one of the 
> classes changed its allowed constructors and another class hasn't 
> picked up the change yet.
>

Confirmed, running make clean followed by make fixes that issue,
> I've noticed another bug: when opening the file you provided, channel
> colors aren't loaded properly. Histograms are drawn with white colors
> and the image display defaults to the [...]

I still have the bug where the top most histogram does not "change the 
image" when moving the triangles (it does change now when changing the 
gamma) in composite mode.  The other histograms now work OK.

Best,

Nico

ChrisWeisiger, 2015-11-05:
Fixed the issue where the first channel of a composite image could not have its contrast adjusted. I believe that just leaves the histogram/channel colors bug I discovered; is that correct?

ChrisWeisiger, 2015-11-05:
Colors are now initialized properly when the file is loaded. This doesn't mean that we load the colors stored in the TIFF; instead, we use the current user's default colors for the given channel name/group. Our Multipage TIFF code does not have (and as far as I can tell, never has had) any mechanism for reading the colors and contrast settings that it writes, and I'm loathe to get bogged down into that when we have an existing, simple method of recovering that information by simply reading the displaySettings.txt file that's generated when the data is saved to disk.

I think that covers all of the bugs. Please re-open if I've missed any!

closed 2015-11-05 by ChrisWeisiger

ChrisWeisiger, 2015-11-05:
Oh, I forgot that gamma isn't used when using lookup tables. That's now also fixed.

nicost, 2015-11-05:
> I've fixed the gamma curve not working in grayscale and the gamma 
> curve becoming so extreme you can't adjust it any more (it now snaps 
> back to a straight curve if you drag it too far). The others will take 
> a bit longer. 

This is much better, but a bit strange.  Would it be possible to - 
instead of snapping back to a straight curve - to stop at an point that 
is not too extreme to still be able to get a handle on the curve (which 
is how 1.4 behaves)?

ChrisWeisiger, 2015-11-06:
On Thu, Nov 5, 2015 at 3:28 PM, Nico Stuurman <notifications@github.com>
wrote:

> This is much better, but a bit strange. Would it be possible to -
> instead of snapping back to a straight curve - to stop at an point that
> is not too extreme to still be able to get a handle on the [...]

I've clamped the gamma to the range [0.05, 50] and made it so that when the
mouse goes outside the "valid" range it snaps to the appropriate extreme.
In 1.4 incidentally, we snapped to 0.0 whenever the mouse was outside the
valid range. I guess this just happened to work out because people usually
want gammas that are <= 1.0, so they didn't encounter the behavior where
they would try to set a very high gamma and suddenly have it snap to 0. It
probably also helps that the 1.4 histograms were larger than they are in
2.0.