micro-manager/micro-manager #90: mm2: Histogram black/white point incorrect in fresh image

nicost, 2015-11-05:
Startup MM with demo configuration.  Snap an Image.  Not that the black and white point triangle in the histogram are both set to 0.  Nevertheless, the image is auto-scaled.  If the Autostretch checkbox happens to have been checked, uncheck it, close the Snap/Live window and snap another image.

ChrisWeisiger, 2015-11-06:
I'm having trouble replicating this; when I snap an image with autostretch turned on, the new histograms show up with their triangles in the correct location, no matter what I do to the autostretch settings and my profile's remembered settings. It's possible that it was fixed by other changes made to the histograms in the last day or two. Could you please verify whether or not it's still happening? Thanks!

nicost, 2015-11-06:
> I'm having trouble replicating this; when I snap an image with
> autostretch turned on, the new histograms show up with their triangles
> in the correct location, no matter what I do to the [...]
I nuked my 2.0 directory and re-installed using the latest nightly. 
Seems to have taken care of the problem.  It is possible that I was 
using an older ImageJ (co-incidentally, the window was opening extremely 
slowly using the previous installation, which also is fixed using the 
current installer).  Will run the installer more regularly in the future,

ChrisWeisiger, 2015-11-06:
Okay, thanks for double-checking.

closed 2015-11-06 by ChrisWeisiger