micro-manager/micro-manager #96: mm2: Quick Access window opens on MM startup, even when it was closed before exiting Micro-Manager

nicost, 2015-11-06:
i.e., how do I avoid it showing up everytime?

ChrisWeisiger, 2015-11-06:
This is intended behavior; it should show up automatically if you have
anything in it (thus, drag all the controls out of it and it won't show up
next time). I take it you would prefer it not be visible unless you access
it through the Tools menu?

nicost, 2015-11-06:
On 11/6/15 12:01 PM, ChrisWeisiger wrote:
> This is intended behavior; it should show up automatically if you have
> anything in it (thus, drag all the controls out of it and it won't show up
> next time). I take it you would prefer it not be visible unless you access
> it through the Tools menu?
I always prefer the application appearing in the same state I left it 
in, i.e. when I closed that window and restart MM, I do not want it to 
show up.  When I leave the window open and restart MM, I do want it to 
show up.  Having to delete the content of that window for it to not 
re-appear on startup, kind of defeats the purpose (it is a lot of work 
to set the things up!).

Also, it would be nice if the buttons would show up in the full screen 
mode (like the Inspector window).

Can the channel color button backgrounds be taken from the MDA window?

The exposure field does not behave as expected (hard to drag into a 
square, does not line up with the square).

ChrisWeisiger, 2015-11-06:
Okay, the window now includes a configuration option so you can decide whether it should open on launch. Options are a) never, b) always as long as there are contents (the current behavior), and c) only if it was open at the end of the last session.

As for your other feedback:

 * For the buttons to show up in fullscreen mode, the window has to have the "always on top" property. I've set it as such for now. It may be that we'll want to make that a configurable property in the future; I'd be cautious about having too many "always on top" windows.

 * Button backgrounds is a notoriously difficult problem as we have approximately zero control over how the program renders buttons (that's all governed by the platform-specific look and feel code). Also the current Presets button is not necessarily only for channels (though that is its most obvious use), and users may have multiple channel groups...bottom line is it's difficult to magically know what color to use, and as I said rendering it properly is more or less impossible. 

This is something we need to tackle; you aren't the only one to request colorized buttons. But I think the best we're going to be able to do is either a) have something that behaves like a button but isn't one (and doesn't look native) whose rendering we can control, or b) have a variety of (possibly procedurally-generated) icons the user can select from to use when configuring the button. Either way I believe the user is likely going to have to manually choose the color of the control themselves.

 * The exposure field is a 1x2 control (it should highlight as such when you drag it into the window). I wasn't able to get it into a single cell; some portion of it would always get chopped off. The same goes for the MDA control.

nicost, 2015-11-07:
Nice!  Having the Quick-Access tool in full screen mode is totally awesome!

ChrisWeisiger, 2015-11-09:
I did a little more finagling of the Exposure Time and MDA widgets, and they now each take up a single cell, at least on Mac. Would appreciate it if you could double-check them on Windows when you get a chance though. Thanks!

ChrisWeisiger, 2015-11-11:
Okay, I think everything in this issue has been taken care of. Please let me know if I missed something; otherwise make a new issue for any additional feedback you have. Thanks!

closed 2015-11-11 by ChrisWeisiger