micro-manager/micro-manager #98: mm2: when opening new dataset, autostretch is active, even when  it is not checked.

nicost, 2015-11-06:
First make sure that autostretch is unchecked (by snapping an image, and unchecking autostretch in the inspector window).  Run an mda with multiple z positions (for instance -3 to 3 with 1 micron steps).  Scroll through the z positions and see notice that the histogram is autostretching even though autostretch is not checked.  Toggling autostretch results in the correct behavior.

ChrisWeisiger, 2015-11-12:
I'm not able to reproduce this now. It's possible that it was fixed when I fixed issue #101. Can you please verify? Thanks!

nicost, 2015-11-12:
I saw it once more with an old dataset, but when I closed it, quit 
Micro-Manager and re-opened it, it behaved fine. You can close, and I 
will file another one if I find a reproducible way to trigger the problem.

Also, had an issue with the Open Recent menu that I could not 
reproduce.  It threw an error when selecting an item from that menu and 
afterwards all menu entries were greyed out.  After opening the file 
directly (File > Open(RAM)), and the using the Open Recent entry, all 
was fine, so that may have been crud left behind in the Profile.  
Nevertheless, would be good to that errors do not trip up the menu.

ChrisWeisiger, 2015-11-12:
Okay, thanks. If you have the logs from that error, I'd like to see them.

closed 2015-11-12 by ChrisWeisiger