micro-manager/micro-manager #99: mm2: bitdepth selection of histogram affects brightness/contrast of display

nicost, 2015-11-06:
Use democamera set to 16-bit.  Take an image at 50 ms exposure.  Check "Autostretch" in the Inspector window.  Toggle the bitdepth between 12-bit and 16-bit (or any other setting).  The histogram display changes as expected, however, the brightness of the display changes unexpectedly, as if it getc clipped at the maximum value of the histogram.

ChrisWeisiger, 2015-12-01:
For background, when you ask ImageJ for a histogram of your image data, it generates a histogram where every single possible value in the image is a separate "bucket". Thus if you have an 8-bit image, you get a histogram with 256 buckets every time, and if you have a 16-bit image you get a histogram with 65536 buckets. 32-bit images are completely out of the question in such a regime if you want any kind of reasonable performance (the bucket would have ~4 billion entries).

In contrast, we have our own histogram-calculation code in 2.0 that takes the desired scale of the histogram into account, and uses 256 buckets at most. I believe this results in a minor but noticeable speed improvement in our histogram code when operating on 16-bit images. Unfortunately, when autostretching, we derive our min/max scale from the histogram -- we have to, because it's the only place where we can perform the "ignore outliers" logic. And the histogram can't tell you information about pixels that are above its "most intense" bucket, which causes the clipping you see.

I'm honestly not sure how to go about fixing this; as far as I can tell the only way to achieve the old behavior is to generate a histogram ImageJ-style and then truncate it, which is excessively inefficient for the vast majority of use cases and will also limit our options down the road for other image pixel types (like 32-bit int or floating point).

How big of an issue do you consider this to be?

nicost, 2015-12-01:
Big.  Changing the bit-depth of the histogram is a tool to zoom into relevant areas of the histogram, but should never change the displayed image.  I understand the technical reasons for this happening, but a better solution should be found (the simplest being to revert to the way ImageJ calculates the histogram, especially since the speed up is only minor).

ChrisWeisiger, 2015-12-01:
I'm going to need some time to think about ways we can attack this. There may be an elegant solution that will work without requiring huge numbers of buckets in the histogram. Mark has also suggested possibly hardcoding the number of buckets to 2^16. Whatever is done will require some reworking of how histograms are drawn, though, so it's not a trivial fix.

nicost, 2015-12-01:
If I understand correctly, the main reason for the current behavior is 
the re-use of the histogram for the "ignore autostretch" functionality.  
Even there, 256 buckets is not enough.  The default 0.1% already needs 
1000 buckets, and I find myself often needing an even lower percentage.  
All of this is quite critical essential functionality, and taking it 
away in 2.0 is not going to be appreciated;)

ChrisWeisiger, 2015-12-01:
Ignore % is handled by linearly interpolating between buckets; it works fine in 2.0.

ChrisWeisiger, 2015-12-03:
Okay, so:

1) In order to properly handle the ignore% functionality, we must generate a histogram whose bins contain the complete range of values represented in the data -- if the histogram doesn't cover the complete range, then we can't know the distribution of pixel intensities at the top end, and thus can't figure out how many pixels to ignore.

This may in fact be solvable through cleverness; in fact "find the 99th percentile without being able to sort the entire dataset" sounds like a CS test problem. It would probably require storing the top (number of pixels * N%) pixels as we calculate the histogram though, and the number of "sorted insert" operations required to do that probably makes it impractical.

2) In order to draw nice-looking histograms, we need to either reduce the bin size as you zoom in, or else have a constant and reasonably-small bin size. Our max zoom level is 4-bit (displays values from 0 through 15) and it'd be good if at that level each intensity had its own bin, so it seems like the "constant bin size" solution would mandate having a bin size of 1. Which is basically the ImageJ approach. Does anyone actually use the 4-bit, 5-bit, or 6-bit zoom levels?

3) It would be nice, long-term, if we had a solution that worked cleanly for 32-bit pixels, though we have no need for this currently.

Does that about sum things up?

marktsuchida, 2015-12-03:
@ChrisWeisiger Sounds right to me.

It should also be noted that computing the percentiles requires a high-resolution histogram, because there can be cases where most of the pixels are concentrated within a small number of intensity values, and you would want the "Ignore X %" to work in such cases. So computing a 65535-level histogram for 16-bit images seems unavoidable, as you mention.

I think use of 4-bit and 5-bit zoom levels is rare but IMO it's worth keeping the support. If all your pixels are in the range 0-15 (4-bit range) the image might not be very useful, but being able to see the histogram for it might be useful for the purpose of troubleshooting, etc.

ChrisWeisiger, 2015-12-07:
This should now be fixed. Please let me know if you encounter any related problems.

closed 2015-12-07 by ChrisWeisiger