micro-manager/micro-manager #100: MM2: add DataProcessor that splits out certain images to a separate Datastore

ChrisWeisiger, 2015-11-09:
This will be important as a hack way to support multiple kinds of images in the same acquisition (e.g. grayscale and RGB, or different sensor sizes). The processor could be configurable to pull out images that are assigned to a specific channel or camera and store them to disk alongside the original acquisition, with or without image display.

ChrisWeisiger, 2016-04-27:
Some thoughts:

 * We've previously had a request to be able to split out each stage position into a separate datastore. Thus some mechanism that has an unknown number of output Datastores will be needed (including the necessary GUI for deciding how to save those Datastores)
 * On a related note, there's not currently any way to get the intended save path of a Datastore without having access to that Datastore (so you can call `Datastore.getSavePath()`)...which Processors do not have. There is `SummaryMetadata.getDirectory()`, but that doesn't have the name of the dataset _in_ that directory.
 * Users may want to select a range of values (e.g. saving the center 3 Z slices), potentially with a "skip" (e.g. only taking odd-numbered channels)
 * Combinations of multiple criteria might be used to decide which images to split out (e.g. only images from every 10th stage position and first channel and center Z slice)
 * Should split-out images still continue through the original pipeline? If not, should the coordinates of original images be "compacted" so that there aren't gaps in the original data?
 * Should it be possible to create a pipeline that takes as its input the output of this processor? That would imply that this is not a traditional DataProcessor and more an integral part of the pipeline (a "fork" in the pipeline)...and also entail a rework of the pipeline GUI as a whole.

Bottom line, we can make this spec very complicated very easily.

marktsuchida, 2020-08-28:
We now have the Image Saver, though it doesn't have the ability to filter images based on coordinates.

I feel that the use cases described here require fundamental enhancements to the processing pipeline (or its integration into acquisition), such as branching and filtering, rather than trying to do from a processor that can only be an item in the linear pipeline.

closed 2020-08-28 by marktsuchida