micro-manager/micro-manager #103: mm2: broken menus

ChrisWeisiger, 2015-11-16:
Repro case:

1) Show a Quick Access Panel
2) Drag the "Run Saved Acq" icon into the grid
3) Cancel the open-file dialog
4) Clear the "failed to configure plugin" error message
5) Select main window, check on the menus: every option is grayed out.

May be related to issue #102; in any event, cancelling a JFileChooser does not have this problem.

ChrisWeisiger, 2015-11-18:
This has now been mostly fixed, in that when the menus break, all of them except for the Help menu now fix themselves. The Help menu on OSX appears to be special in some way that I can't determine, and I haven't been able to figure out a workaround that fixes it. 

This source code may potentially be relevant:
http://cr.openjdk.java.net/~michaelm/7113349/7u4/2/jdk7u-osx/new/src/macosx/classes/com/apple/laf/ScreenMenu.java.html

Upgrading to Java 8 may also make a difference.

Note that the original repro case for this bug doesn't work because the Quick Access Panel now more gracefully handles configuration errors, but this can easily be "patched in" by removing the "if (config == null)" check in QuickAccessFrame.dropPlugin().

nicost, 2020-04-14:
Can not reproduce

closed 2020-04-14 by nicost