micro-manager/micro-manager #105: mm2: easily updatable presets

ChrisWeisiger, 2015-11-19:
It's fairly common for users to have a channel preset that incorporates an intensity value (e.g. GFP + 75% laser intensity). When they adjust the intensity, they want it to be remembered the next time they select that channel. Currently this intensity is "baked in" to the preset, and updating the preset is laborious; instead we should detect when a property that is part of a currently-active preset has been changed, and allow the user to apply the change to the preset (e.g. add a "!" button to the preset table that can be clicked to save the new setting).

marktsuchida, 2015-11-21:
Rather, it is impractical, though unfortunately not unheard of, for users to have a channel preset that incorporates an intensity value.

I'm not entirely convinced that this is more than just a proposal to make µManager work the way Elements does, arising from an attempt to understand µManager in Elements' terms.
Changing the configuration groups to behave like this just doesn't fit in with the existing data and interface model (not to mention the current implementation). Elements might work that way, but their underlying model (their rough equivalent of µManager's configuration groups) is presumably quite different, so copying the concepts doesn't make sense.

Also note that laser intensities in µManager are usually separate properties for each laser channel. There are other illumination devices where a single property does control intensity of all channels, as in the case of a white LED; in that case, it is indeed sometimes necessary to set the same intensity property differently for each channel, and this is where µManager's current interface is inconvenient.

For such (white-LED-like) cases, perhaps we could add something we might call "tied properties". These would be 1 or more properties associated with a configuration group (not preset), and no values would be saved in the hardware configuration. Their behavior would be similar to the behavior of the camera exposure when "Sync exposure between main and MDA windows" is enabled (except that it doesn't have to be the current channel group). Their values never need to be saved to disk, though they could potentially be saved to the user profile.

An example would be a configuration group (intended for use as the channel group) that switches an excitation filter wheel in front of a white LED. The LED intensity property would not be in the configuration group, but tied to it. Separately, there could be a single-property configuration group for the LED intensity (which appears as a slider in the GUI). The slider would then control the intensity for the currently selected channel (if any).

ChrisWeisiger, 2015-11-23:
I wouldn't get too hung up on Elements' implementation. It seems clear that the underlying goal is "I want to be able to conveniently get back to the same imaging settings as what I used last time." The fact is that currently in order to link intensity to channel, you have to put them both in the same config group, but that is inconvenient to modify when your preferred settings change.

Currently we remember the exposure time associated with the channel; other settings like illumination intensity or ND filters or whatever ought, in principle, to be able to be remembered. A potential UI for this then could be a dialog that allows the user to associate a certain set of properties with the configuration group. When a preset from that group is selected, we update those properties based on which preset was chosen and what we remember the last settings to have been for that preset. We could use the same mechanism for the exposure time (though it's not exactly a traditional property). 

This sounds somewhat similar to your "tied properties" suggestion, and that could be what we call it in the UI.

The problem becomes that we need some mechanism for informing whatever code is responsible for this whenever a preset is chosen. For ease of implementation, I would prefer to do this in the Java layer (where we have ready access to the user profile system), but that wouldn't play well with the acquisition engine, would it? And unfortunately it looks like the core callback system doesn't include an event for when a preset is selected from a config group. There's the configGroupChanged callback, but as far as I can tell that's for when a config group is modified by adding a new entry, not for when an existing entry is selecetd. I could be wrong though.

marktsuchida, 2020-08-28:
I do feel there is some merit to this idea, provided that it can be implemented in a way that does not confuse any of: new users, existing users, or the config file model. A clean data model will need to be established first (which may include first cleaning up the rough edges that config groups currently have).