micro-manager/micro-manager #107: MDA unresponsive when setting 0 exposure in single-channel stream acquisition

omsai, 2015-11-21:
Reproducible:  Always

Steps to reproduce:

1. The minimal hardware configuration needed is a DemoCamera with its camera and shutter checkboxes.
2. Enable "Time points" and "Channels" checkboxes in MDA.
3. Select a Channel group, click "New" to add one, and only one, channel to the MDA channel list (the bug will not manifest if there are >= 2 channels).
4. Enable streaming by setting "0" exposure in all 3 places, namely in the exposure of that new channel (hit <Enter> to save), in the Interval of "Time points" in MDA, and in Exposure [ms] of the main Micro-Manager window.
5. Choose a large number of time points like 10,000.
6. Hit "Acquire!"

Result:

1. Acquisition shows no window for several minutes of waiting.
2. Clicking "Stop" in MDA window and then "Acquire!" gives message window "Cannot start acquisition: previous acquisition still in progress."
3. One has to restart Micro-Manager (without hardware or PC restart) to return to normal.  Not that we used hardware in this test case.

Workarounds:

1) Either set a small exposure in the channel list of MDA, or
2) Uncheck "Channels" in MDA, or.
3) Set a small exposure in "Exposure [ms]" of the main MM window.

Versions affected: 1.4.20, 1.4.22, 2.0-beta2
System: Windows 7 64-bit

ChrisWeisiger, 2015-11-23:
I said in the email thread that I wasn't able to reproduce this in 2.0. Turns out that I wasn't setting a large enough number of timepoints -- 500 won't reproduce, but 5000 will (on my laptop). I do still eventually get an image window and the timeout error dialog, but there's a lengthy period during which no activity is visible. I know Mark has promoted the idea of showing an image window or at least *something* during the time in an MDA when the MDA has started but no images are available; that may be a "fix" in this situation.

ChrisWeisiger, 2016-01-29:
I've added a brief popup window that shows up when you start an acquisition and images don't immediately start showing up. It'll show from 1s after starting acquisition until just before the image display window is displayed. That should provide the necessary feedback. 

Otherwise, are we aware of any other reproducible problem here in 2.0?

marktsuchida, 2016-01-29:
But we shouldn't ever be getting timeout errors from these operations.

Here's what I see:
- Standard demo config, 100 frames, 0 interval, one channel selected (or no channels), exposure 1 ms
- Start the MDA
- I get about 2 to 18 images (variable) in the image window and a timeout error
- I don't see this issue in 1.4 (even with 0 ms exposure) -- there is a delay when starting a long time series, but no timeout error.

I'm sure the frame rate and count needed to reproduce this is machine-dependent, but there seems to be some problem triggered by images coming in at a high rate.

The timeout error occurs in the acquisition engine, which is weird, because I see at least a few frames remaining in the sequence buffer after the error. It would seem that some behavioral difference crept into the acquisition engine between 1.4 and 2.0, or else 2.0 interacts with the engine in some subtly different way.

The timeout used is sent from mmstudio to the acquisition engine as `SequenceSettings.cameraTimeout`. If this value is set smaller in 2.0, it would explain the symptom.

Which leads to the finding that SVN r15785 (dcfd9db6928edcab3b4433f1d6e0876caf17c22e) never made its way into the mm2 branch. Pull request coming shortly.

PR #159 (2016-01-29): Cherry-pick fix for MDA timeout at short exposures
Fixes #107.
closed 2016-01-29 by marktsuchida