micro-manager/micro-manager #110: Plugins in MMJ_.jar get loaded multiple times when running from NetBeans

marktsuchida, 2015-11-25:
Originally reported here: http://micro-manager.3463995.n2.nabble.com/Plugin-is-loading-multiple-times-tp7586337.html
"It looks like every plugin is being loaded once for every jar in the mmplugins directory (and every jar in the mmautofocus directory)." (as summarized by @ChrisWeisiger ).

ChrisWeisiger, 2015-11-25:
This appears to be down to something to do with class loaders and/or SciJava's plugin discovery mechanism, as we are somehow discovering jars that should be invisible to the current class loader. It's possible that SciJava is doing something unusual to enable discovering plugins when run from within NetBeans (and/or Eclipse). It's also possible that NetBeans is doing something unusual to enable itself to work at all, which interacts poorly with our runtime plugin discovery. Suggest next course of action is to contact the SciJava folks and see if they have any ideas.

marktsuchida, 2015-11-25:
* SciJava has special code for Eclipse support, but I have checked that it doesn't get executed when running from NetBeans.
* We call `org.scijava.plugin.PluginIndex.discover()`, which in turn calls `org.scijava.plugin.DefaultPluginFinder.findPlugins()`. The latter method uses `org.scijava.annotations.Index<org.scijava.plugin.Plugin>` to iterate over all `META-INF/json/org.scijava.plugin.Plugin` resources (and entries therein) for the given class loader (in this case our `org.micromanager.internal.pluginmanagement.PluginFinder$PluginLoader` instance).
* That enumeration invokes the class loader's `getResources(String)` method, which is implemented in the ancestral `ClassLoader` class. `ClassLoader.getResources()` returns the combined results from the parent's `getResource()` and the current class's `findResource()` methods.
* According to `ClassLoader` javadoc, it is intended that subclass override the `findClass()`, `findResource()`, and `findResources()` methods, rather than `loadClass()`, `getResource()`, and `getResources()`.
* The reason `MMJ_.jar`'s `META-INF` gets searched is because all resources of the matching name in the ancestral class loaders are being enumerated (not sure why this doesn't happen outside of NetBeans).
* So a naïve fix would be to override `getResources()` in `org.micromanager.internal.pluginmanagement.PluginFinder$PluginLoader` to not look at the parent class loader's resources. But that, of course, is incorrect because we need plugins to be able to load and enumerate their regular resources.
* So one way to fix this is to make the class loader ignore its parent _only_ while plugin discovery. Ugly as this may be, it seems simpler than any other fix.

Created pull request #111 that implements this "stateful class loader" approach. @ChrisWeisiger, please merge if you see no issue with this.

ChrisWeisiger, 2015-11-30:
Pull request is in, so this issue can be closed.

closed 2015-11-30 by ChrisWeisiger