micro-manager/micro-manager #112: mm2: WIndows open very slowly (in Java 1.8.0_66)

nicost, 2015-11-25:
Opening of an image window takes about 2 seconds.  There is a nice animation going on, with the window growing first top to bottom, then left to right.  Likely linked to the specific Java version, but I need to use Java 1.8 now to be compatible with ClearVolume.

ChrisWeisiger, 2015-11-25:
I know that Java 7 has horrible display performance on OSX at least. Haven't tried Java 8 though. Are you on Mac or Windows?

nicost, 2015-11-25:
Mac OSX 10.10.5

nicost, 2015-12-02:
Should add that I only see the slowness when running from NetBeans.  Running it directly seems fine, so probably not very important.

ChrisWeisiger, 2015-12-02:
Okay, that's very useful to know, thanks!

ChrisWeisiger, 2015-12-22:
Can confirm this on my laptop when running outside of NetBeans. This launcher invocation has the slow image windows:

    /Library/Java/JavaVirtualMachines/jdk1.8.0_65.jdk/Contents/Home/bin/java -Xmx1024M \
       -cp mmplugins/dependencies/*:ImageJ.app/Contents/Resources/Java/ij.jar ij.ImageJ \
       -XX:MaxDirectMemorySize=1000G \
       -Dmmcorej.library.loading.stderr.log=yes \
       -Dorg.micromanager.corelog.dir=/tmp \
       -jar "/Users/chriswei/apps/ImageJ/ImageJ.app/Contents/Resources/Java/ij.jar" \
       -eval 'run("Micro-Manager Studio");'

while this one does not:

    /Library/Java/JavaVirtualMachines/jdk1.8.0_65.jdk/Contents/Home/bin/java -Xmx1024M \
       -XX:MaxDirectMemorySize=1000G \
       -Dmmcorej.library.loading.stderr.log=yes \
       -Dorg.micromanager.corelog.dir=/tmp \
       -jar "/Users/chriswei/apps/ImageJ/ImageJ.app/Contents/Resources/Java/ij.jar" \
       -eval 'run("Micro-Manager Studio");'

Since both use Java 8, with the only difference between the two being the -cp parameter (which was used to get the ClearVolume plugin to run), I expect that the problem is due to us using the wrong jar somehow, or something along those lines.

ChrisWeisiger, 2015-12-22:
Okay, this is weird: if I modify the `-cp` parameter to include `mmplugins/dependencies/*jar` instead of just `mmplugins/dependencies/*` then I get the fast window opening. The only files in the dependencies directory end with `.jar`, so the two should be logically equivalent. There's no hidden files in the directory either. I'm confused.

nicost, 2015-12-22:
When you add the ".jar" to the classpath, you will no longer load the 
jars (you can probably check by running the ClearVolume plugin, it most 
likely will fail because it can not fun dependencies.  Just "*" will 
load the jars files.  See 
http://docs.oracle.com/javase/6/docs/technotes/tools/windows/classpath.html, 
and 
http://stackoverflow.com/questions/1237093/how-to-use-a-wildcard-in-the-classpath-to-add-multiple-jars.

So, one or more jars in mmplugins/dependencies is causing this 
behavior.  It would be great if you could triangulate to find out which 
one (so that we can figure out how to fix it).

ChrisWeisiger, 2015-12-22:
Ah, I see. Java's parser doesn't support proper globbing, just `*` as a special case.

The bad jar is ij-1.49v.jar. Removing it fixes the problem. For reference, I'm running against ij-1.49h in my local install.

marktsuchida, 2015-12-22:
Just a thought, one thing you might check is the Java system properties (which `ij.jar` might be setting). There may be some parameters about rendering, graphics acceleration, or UI. See, e.g., the dump in our Problem Reports.

But since this is presumably about opening a window, not drawing its contents, it's also possible that it's a real code difference between the `ij.jar` versions, of course. It might be worth checking what the most recent 1.50 version does, just for future reference....

BTW, did you mean 1.48v (not 1.49v)? 1.4 currently ships with 1.48v, so that would more likely be left over in your build tree.

nicost, 2015-12-22:
I checked 1.50e (which is the newest in maven) and it has the same 
problem.  Apparently, Wayne put a lot of effort into making it 
compatible with java 8, so it seems prudent to figure this issue out.  I 
just tested 1.50e and it has the slow windows phenotype with MM, but 
open windows fast by itself, so it would be good to figure out what is 
causing it and fixing (either in the MM or ImageJ code).  B.t.w., it was 
1.49v (which I included through the pom.xml in the MMClearVolume 
plugin.  The change happened between ImageJ version 1.49j (good) and 
1.49k (bad).

ChrisWeisiger, 2016-04-14:
Another thing to be on the lookout for is memory leaks. With ij-1.49h and Java 8, I get a memory leak with each display I close. It's definitely possible that a newer version of ij.jar will fix that; just adding to the list of things we need to check when we get around to working on the whole Java 8 upgrade...

kmdouglass, 2017-01-31:
Hi @ChrisWeisiger ,
I just came across this thread while looking into some issues concerning my system's memory usage with MM2.

Is there any update as to whether MM2 and Java 8 is leaking memory when a display is closed? I'm noticing that the in-use memory on my system (as read out by Windows's Task Manager) is growing every time a complete Multi-D acquisition is saved to RAM and then has its window closed. To be fair, I'm not sure whether this memory is actually in use or just not being released by the JVM back to Windows.

I'm writing a post to the mailing list now with more details; I just want to be sure I'm not trying to identify an already known-problem.

marktsuchida, 2017-01-31:
@kmdouglass Nothing known at the moment. I am working on fixing a whole bunch of display bugs, though. Also note that the memory shown in Task Manager does not necessarily indicate a memory leak.
Hopefully we'll be able to get everything working smoothly in Java 8 this spring or summer.

nicost, 2020-01-31:
Newer Java versions and lots of internal work seems to have solved most of the issue mentioned here.

closed 2020-01-31 by nicost