micro-manager/micro-manager #114: mm2: Can not achieve zoom level 100%

nicost, 2015-11-30:
Snap an image. It opens at 75% zoom level.  Click the "+" button.  Zoom level now is 150%.  Click the "-" button.  Zoom level is again at 75%.  I see no way to reach the highly desired 100% zoom level.  100% maps one pixel in the image to one pixel on the screen and is a very useful zoom level, so should always be reachable.

ChrisWeisiger, 2015-11-30:
Strange. What are the dimensions of the image (in pixels)? And what is your monitor's resolution?

I'm unable to achieve a zoom level of 75% in my tests, but I can certainly reach 100%, so there must be something subtle going on here.

ChrisWeisiger, 2015-11-30:
Oh wait, please disregard my questions. I've found the cause; fix incoming...

ChrisWeisiger, 2015-11-30:
And this is now fixed. Our zoom buttons had been doing exclusively power-of-2 zooming instead of allowing the 75% and 150% zoom levels that ImageJ did; ImageJ's own zoom tool was not thus constrained and could get you to zoom levels that our zoom buttons then would get "stuck" on. Now we use basically the same zoom logic that ImageJ does.

closed 2015-11-30 by ChrisWeisiger

nicost, 2015-11-30:
Yes!  Thanks!