micro-manager/micro-manager #118: mm2: Each Inspector tab needs to be clicked once before it becomes active

nicost, 2015-12-05:
First click does nothing.  Afterwards all clicks have the expected effect.  Each of the tabs have to be "woken up" before things start working as expected.

ChrisWeisiger, 2015-12-07:
Strange, I'm not able to reproduce this. Does it happen on Windows? Mac? 

Looking at the logic (in the MouseListener in InspectorPanel.addPanel()) the only things I can think of that would result in the behavior you're seeing are a) the panel header is not receiving the first mouse event for some reason, or b) the panel is returning false from its isVisible() call despite having been set visible immediately prior.

nicost, 2015-12-07:
> Strange, I'm not able to reproduce this. Does it happen on Windows? Mac?
>

Only tested on Mac.  It is completely reproducible, and independent of 
the plugin that I am working on.  All with Java 1.8.

> Looking at the logic (in the MouseListener in
> InspectorPanel.addPanel()) the only things I can think of that would
> result in the behavior you're seeing are a) the panel header is not
> receiving [...]

The first click (and all others) produce the following output:

2015-12-07T10:13:17.890535 tid0x12f557000 [IFO,App] [UI] JPanel clicked 
in InspectorFrame.

So, something receives the click.  All subsequent clicks have the 
expected result, just not the first one.

ChrisWeisiger, 2015-12-07:
The "header" object (gray rectangle with the panel's name in it) is a JPanel...but so are several other objects in the UI hierarchy. 

This seems most likely to be a change in behavior between Java 6 and Java 8. I've pushed a change that *might* fix this, so please re-test at your leisure.

nicost, 2015-12-29:
I no longer experience this, so you must have fixed it!  Thanks!

ChrisWeisiger, 2015-12-29:
Awesome, thanks for the confirmation!

closed 2015-12-29 by ChrisWeisiger