micro-manager/micro-manager #124: MM2: Wrong color in live view

ChrisWeisiger, 2015-12-21:
Snap image with channel 1. Switch to channel 2. Snap new image. Color of displayed image does not change. Ideal behavior would be to recognize that the channel has changed and sub in the new color. This may require some changes to the Snap/Live data management since I believe it does not currently set channel names.

ChrisWeisiger, 2015-12-22:
Can confirm this will require us to propagate through the channel name(s) to the summary metadata, as they aren't currently actually available anywhere in the snap/live view. This becomes more complicated in multi-camera cases, and of course is even more problematic if a data processor in the pipeline is creating additional channels (e.g. the Split View processor). This is going to require some thought.

marktsuchida, 2015-12-22:
I think having Summary Metadata for Snap/Live makes a lot of sense. It would need to be constructed newly for each snap, or each live run. I guess they would be treated as a single-channel acquisition that nonetheless has the Channel coordinate (when Core-ChannelGroup is set; otherwise they can remain channel-less).

But there's another situation that is related to this: what if the user adds a bunch of snaps to an Album, switching between channels in the process? Ideally, each image would be displayed in the correct color when the display mode is set to Color. One idea would be to update the Album so that its summary metadata looks like a multi-channel acquisition, although I'm not sure what all the ramifications of this will be.

Once that is done, can the processors just take care of themselves, by transforming the SummaryMetadata (as SplitView already does)?

ChrisWeisiger, 2016-01-05:
Do we consider this to be a 2.0 release blocker?

marktsuchida, 2016-01-06:
I would consider it a release blocker, but I would be okay with releasing without this enhancement if the Snap/Live window can be pinned to Grayscale (or, more accurately, Color and Composite are disallowed in the Snap/Live window, but without affecting the default for the next MDA (etc.) window).

ChrisWeisiger, 2016-01-06:
We can't lock the Snap/Live view in Grayscale because that breaks multicamera setups. I guess that means this is a release blocker.

ChrisWeisiger, 2016-01-06:
This is now fixed; we set proper summary metadata in the Snap/Live datastore, and reset the entire display if the user changes channels. The new display then pulls different channel colors based on what it finds in the summary metadata. This is somewhat inelegant (if you change channels, then the window disappears and reappears, with associated UI lag), but I believe that the Snap/Live view is the only situation in which we ordinarily expect that the channels can change on the fly, so I did not feel it was appropriate to add logic to the display layer to detect changes in the channels and pull new channel colors to suit.

closed 2016-01-06 by ChrisWeisiger