micro-manager/micro-manager #129: mm2: SetDispaySettings api call throws null pointer exception

nicost, 2015-12-29:
Can be reproduced with the following script:
```
store = mm.data().createRAMDatastore();
display = mm.displays().createDisplay(store);
dsb = display.getDisplaySettings().copy();
display.setDisplaySettings(dsb.build());
```
And is caused by the use of the 'canvas_' variable in the SetDisplaySettings function without checking for null.  The variable is only set when an image is actually displayed.  However, as a script writer, it is quite logical to want to set the display settings before displaying images (and it becomes nasty to keep track of when one adds the first image to then set the display settings).

ChrisWeisiger, 2015-12-29:
Yes, this should not be happening. Thanks for the report.

ChrisWeisiger, 2016-01-05:
This should now be fixed. Thanks for the report!

closed 2016-01-05 by ChrisWeisiger