micro-manager/micro-manager #133: mm2: Mouse-moves-stage interferes with panning

marktsuchida, 2016-01-05:
When mouse-moves-stage is on and the ImageJ tool selection is the Hand tool, but the window is not displaying the full image (i.e. it is zoomed in), dragging on the image appears to do some weird mixture of panning the image (in software) and panning the field of view (by moving the XY stage).
(Haven't tested if 1.4 handled this correctly.)

ChrisWeisiger, 2016-01-05:
The 1.4 behavior appears to be that the stage only moves if the user is dragging the view while at the edge of the image.

ChrisWeisiger, 2016-01-29:
What is the desired behavior here? Should we refuse to move the stage if you're panning the view when zoomed-in?

marktsuchida, 2016-01-29:
Current behavior sounds reasonable. I felt it behaved oddly when testing with a real stage and sample, though I could have been mistaken. Let’s test this on hardware before closing.

marktsuchida, 2020-11-26:
As of today, the original description (mixture of image display scrolling and stage movement) still holds.

This should be fixed but is probably not super urgent.