micro-manager/micro-manager #137: mm2: Bugs in linking histogram settings

marktsuchida, 2016-01-06:
Two bugs, maybe not related to each other.

For example, run an MDA with time points and channels (dataset 1). Run it once more to create a second dataset (dataset 2).
Now, in the inspector for dataset 2, link the histogram settings for the first channel to dataset 1.

Bug 1: Initially, dragging the intensity range (triangle handles) does not change the intensity range for the dataset 1 window. (Fixes itself after the dataset 1 window is clicked once.)

Furthermore, open a second inspector and attach one inspector to each window.

Bug 2: The link button for the channel 1 histogram is not highlighted (green) in the second inspector. (It does correctly get highlighted if the link is performed after opening both inspectors.)

ChrisWeisiger, 2016-01-06:
I was initially able to reproduce the first bug, but now cannot. The second bug is now fixed.

ChrisWeisiger, 2016-01-11:
This is now fixed.

closed 2016-01-11 by ChrisWeisiger