micro-manager/micro-manager #143: mm2: saving/loading 1.4 comments is broken

ChrisWeisiger, 2016-01-08:
MM2's multipage TIFF format currently only saves the summary metadata. It needs to save the per-frame metadata as well. And it needs to load the per-frame metadata from 1.4 images correctly and reconstruct images appropriately.

marktsuchida, 2016-01-09:
See also: #141.

In 1.4 TIFF stacks, comments were _not_ saved in either the summary metadata JSON at the beginning of the file, or the per-image metadata JSON linked from each TIFF IFD. They were only saved at the end of the file in their own JSON string.
This was done by explicitly removing the Comment key from the summary metadata JSON before writing it to file.

Note that we need to also load comments from 1.4 single-image TIFF format.

In 1.4, the behavior of comments was not cleanly defined and problematic: comments could be edited any time before saving an in-memory image, but only before acquisition for on-disk images. But the UI did not give any indication when an uneditable comment got edited (it would just silently loose the edits).

Comments saved in 1.4 and loaded by 2.0 need not be editable.

How should we handle comments saved by 2.0?

For the file format, there are 3 possibilities (under the principle that the TIFF file should not be modified once the data is finalized): (1) put comments in the (summary and per-image) metadata, (2) put comments in a separate JSON file, or (3) both.

Approach (1) has the advantage that comments are not easily lost, but will prohibit editing comments after an acquisition has finished. (2) will allow us to support editing comments any time (even after reopening the image), but the user may lose comments if the external file is not copied together with the image. (3) will ensure that at least the "original" comments are preserved with the file while also allow editing, but has the potential disadvantage that the behavior might be confusing to users.

Approach (1) is closest to the current state of the mm2 branch, and I think the disadvantage of (2) is greater than the disadvantage of (1). Also, (1) does preserve the 1.4 behavior (albeit not clearly intentional) that comments never change after the acquisition session. (3) can be implemented in the future whether we choose (1) or (2) for the short term.

But there is a further problem with approach (1): in disk-backed datastores, the file format does not (realistically) allow modifying any metadata JSON once it is written (summary at beginning, per-image with each image). So if we do have comments in the metadata, they will need to be limited to "pre-acquisition" comments: summary comments entered before starting the acquisition, or per-image comments that are generated (probably not interactively) before the image gets inserted into the datastore. And even with RAM-backed images, it is very unnatural that the user suddenly cannot edit comments when the acquisition finishes (and is thus frozen).

I'm afraid this completely rules out the route of saving comments as part of the metadata JSON (something 1.4 never did, probably for this reason).

Given this issue, we might also consider approach (1a): write comments in the TIFF stack file, but in a separate JSON string at the end of the file. This is very similar to 1.4, except that the JSON format may need to be different. With (1a), we can allow editing of comments even after the datastore is frozen (but not after reloading a saved file).

Approach (1a) will allow us to support the following behavior:

- User can enter a summary comment before, during, or after running an MDA, regardless of file format (single-image, stack) or storage (disk-backed, RAM-backed), until the datastore is closed
- User can enter a per-image comment during or after an MDA, regardless of file format or storage, until the datastore is closed
- User cannot enter or edit comments on datastores loaded from saved files, regardless of file format or storage (the UI should make this clear; see #142)

(Note that, strictly speaking, this requires that RAM-backed datastores turn into disk-backed datastores upon "saving" _without_ loosing the ability to edit comments, despite being frozen. If that is not easy, a compromise implementation where RAM-backed datasets stop allowing edits upon "saving" is also good enough, I think.)

(Note also that this idea involves keeping the stack TIFF file open for writing beyond when it is frozen; to ensure data integrity (e.g. when the next acquisition crashes), we should take care to flush a valid, complete file upon freezing, even if comments may be overwritten later.)

So the best I can currently think of is (1a). But this does mean that comments will need to be treated differently from metadata throughout the data structures -- something that was rather painful in 1.4. If this turns out to be impossible given the mm2 API design, we'll need to think harder.

[We could consider a (3a) (= (1a) + (2)) in the long term, to allow amending comments at a later time. But to avoid confusing users, it would be best if the "original" comments (stored in the TIFF and uneditable after reopening) and "editable" comments (stored in `comments.txt`) be kept explicitly separate in the UI (otherwise, the user might be confused by the fact that some but not all comments are retained when `comments.txt` is lost, or, worse, that some comments silently revert to the original version). Given the complexity that this introduces, maybe we should never do it. I'm a little torn on this, though, because it is indeed nice to have a way to indicate corrections to erroneous entries made in the comments at the time of acquisition.]

ChrisWeisiger, 2016-01-09:
I'll process this more fully later, but thanks for the analysis, Mark!

ChrisWeisiger, 2016-01-11:
For the short term (initial 2.0 release), from what I can tell we must do the following: 

1) Ensure that loading of 1.4 datasets includes their comments correctly.
2) Ensure that 2.0 is capable of all comment handling that 1.4 can do.

Ideally we also at the very least don't interfere with future work to make comments easier to use. It looks like we're probably going to need some mechanism for a separate "comments block" in the file which is read whenever a given image record is read, so that the appropriate comment, if any, can be inserted into the image. We may as well store these comments the same way 1.4 is storing them, at least for the time being.

ChrisWeisiger, 2016-01-11:
2.0 now correctly loads 1.4 comments and writes comments in a fashion that it can itself read. 1.4 cannot properly load 2.0 multipage TIFFs, but I don't believe that's necessarily a release blocker.

ChrisWeisiger, 2016-01-27:
I believe this issue is now resolved; comments are stored in a separate comments.txt file in 2.0 using the new Annotation system, and when 2.0 loads a 1.4 file, it extracts any comments it finds and copies them to that file.

closed 2016-01-27 by ChrisWeisiger