micro-manager/micro-manager #147: mm2: add Datastore variant that allows image deletion

ChrisWeisiger, 2016-01-22:
Some third-party code may want the ability to delete images from a Datastore. However, our standard Datastore should not allow deletion (it should be write-once, read-many). So we should probably add a variant Datastore that does allow this and is limited to specific types of Storage (to begin with, StorageRAM) that can implement the desired behavior. We should also add appropriate events so clients can be informed when images are deleted or overwritten.

Then we should use this new Datastore type for the live view, which currently relies on the fact that StorageRAM responds to repeat insertions at the same coordinates by overwriting the old image.

ChrisWeisiger, 2016-01-22:
Proposal: make a new interface (for lack of a better name at the moment call it DatastoreB) which inherits from Datastore. DatastoreB behaves just like Datastore, except that a) it allows putImage() calls to overwrite pre-existing images at the same coordinates, and b) it provides deleteImage() and clearAllImages() methods. We will need an implementation (DefaultDatastoreB) for this interface as well.

Add an inheritor of the Storage interface (call it StorageB) for Storage classes that can support the above behaviors. Have StorageRAM implement StorageB; our disk-based Storage options can continue to implement Storage instead.

Add a createRAMDatastoreB() method to DataManager, which returns a DatastoreB backed by a StorageRAM.

Add events to notify subscribers when images are deleted or overwritten or the Datastore is cleared.

Maybe call this an ErasableDatastore?

marktsuchida, 2016-01-23:
Another name candidate (which I like better): `RewritableDatastore`.

We need to add a `throws DatastoreNotRewritableException` to `Datastore.putImage()` before 2.0 release (but note that the effect of putting to an existing coordinate is presently undefined -- unless we can forbid it now and somehow work around it in Live view, which would be much more ideal).

ChrisWeisiger, 2016-01-25:
The basic implementation is surprisingly straightforward, but there's one big issue we need to sort out. If we by default use standard Datastores instead of ErasableDatastores (or whatever we decide to call them) in µManager, then we cannot modify image comments in *any* Datastore (not just Datastores whose `freeze` method has been called), because modifying a comment requires replacing the image in the Datastore with a new one with updated comment data.

It seems likely that incorporating these more-mutable Datastores into µManager will require a revamp of how we handle comments, which is something we've been meaning to tackle anyway. I think that's probably a big enough change that I don't want to try to squeeze this particular feature into the first 2.0 release, though. I'll create a separate issue for updating our comments system.

ChrisWeisiger, 2016-01-27:
This is now done, and as issue #148 is now resolved, we can resolve this one as well.

closed 2016-01-27 by ChrisWeisiger