micro-manager/micro-manager #157: mm2: RGB histograms: drawing issue of limit bars

marktsuchida, 2016-01-29:
The broken-line vertical bars indicating the histogram limits should be erased when switching between R, G, B. Currently the upper limit bar seems to linger. Seen on Windows.

ChrisWeisiger, 2016-01-29:
To be clear, your argument is that for something like this:

http://i.imgur.com/NLUWWiZ.png

the red and green vertical lines should not be drawn because their components are not active?

marktsuchida, 2016-01-30:
Actually, no. I was confused by the behavior when switching Autostretch on and off.

When autostretch is turned on, clearly all 3 components are autostretched in the displayed image, but the limit bars remain where they happened to be before autostretch was turned on.

ChrisWeisiger, 2016-02-02:
This is now fixed.

closed 2016-02-02 by ChrisWeisiger