micro-manager/micro-manager #158: mm2: Live view is significantly slower than in 1.4

marktsuchida, 2016-01-29:
Tested with a 2048x2048 camera on Windows. But maybe try DemoCamera at 2048x2048, 20 ms exposure, FastImage enabled.

Display frame rate is about 2/3 of 1.4 when histogram update is disabled ('slow' in 1.4). With histogram updated every frame, 2.0 is even slower (this is when the problem becomes potentially severe). Autostretch adds significantly to the slowness.

A better controlled comparison should be made first.

ChrisWeisiger, 2016-01-29:
Observed display framerates (not data rates) with 1 2048x2048 DemoCamera running live mode with a 10ms exposure time, FastImage enabled, autostretch on, on my Macbook:

1.4:
Histograms every image: 2.3 - 2.5FPS
"Slow Hist" checked: 2.5 - 3FPS

2.0:
Histograms every image: 2.4FPS
Histograms once per second: highly variable, from 3 - 8FPS (if FastImage is disabled, then there's a noticeable hiccup once per second)
Histograms never: ~22FPS
Histograms every image with a 128x128 ROI: 15 - 20FPS

Looking at NetBeans, an awful lot of time is spent in the ContrastCalculator.calculateHistogram() method, almost entirely in its own code (i.e. not methods it calls). It has to iterate over every single pixel in the ROI, so it's not that surprising that it's slow, but it could probably be made more efficient. In particular, having different versions of the inner loop that work for different data types (byt[] vs. short[] vs. multi-component byte[]) would probably be good so we can avoid iterating over each potential component of each pixel.

Of course, we could also introduce a sampled histogram calculation mode where we only look at every 4th or 8th pixel or the like. That would be an easy speedup, at the cost of accuracy.

marktsuchida, 2016-01-30:
Windows 7 x64, Core i5 Haswell.
2048x2048 fast image, 10 ms exposure, autostrech on, 33% zoom.

1.4:
Histo every image: 29-31 fps
Slow hist: 43-47 fps

2.0:
Every img: 2-3 fps
1 per sec: 2-6 fps (variable)
Never: 22-27 fps
Every img, 128x128 ROI: 22-27 fps

(Display fps took quite some time to equilibrate when switching between every image and never - which makes me wonder why it fluctuates so much if it has such a long time constant....)

So 2.0 behavior is consistent with your observations, but 1.4 is not. It would appear that the histogram computation is much faster in 1.4 (ImageJ1 code). Looking at the code, this is not surprising. Perhaps the inner loop fails to get JIT-compiled in 2.0.

I think having 3 versions of the loop over pixels is a good idea, for now. (Some day we'll be using ImgLib2, I hope.)

ChrisWeisiger, 2016-02-01:
I've made a number of optimizations that should ameliorate this problem:

 * Different pixel types have different functions for calculating histograms
 * Different functions are used for ROIs vs. non-ROI images (allows us to skip an if statement for each pixel)
 * No more "manual double buffering" in the canvas paint logic, which saves an extra drawImage() call (at the cost of making the Export Movie Dialog slower, but that seems fine to me). 

NetBeans on my laptop now says that the vast majority of our time (~75%) is spent in ImageJ code actually painting the image to the screen. Our ability to effect improvements there is rather limited.

ChrisWeisiger, 2016-02-02:
Tested on Windows; histogram calculation is still using an unacceptable amount of time on that platform. So this isn't really properly fixed yet.

ChrisWeisiger, 2016-02-03:
I wrote up a simple test program that times our histogram calculation method and compares it to the method ImageJ (and therefore 1.4) uses. It turns out that the big cause of the slowdown is our index calculation: `int index = (y * 512 + x) * stride + component`. We do this calculation once per pixel. By comparison, ImageJ calculates `y * 512` once and then adds 1 to it for each pixel. ImageJ's method only works when you have a single-component image (and a stride between pixels of 1, but that should always be true of single-component images). 

So I suspect our fix for this needs to be to recognize when we have a single-component image and use a separate histogram calculation method for those situations. Our RGB histograms will still need to use the slow method.

ChrisWeisiger, 2016-02-03:
This is now fixed; I'm getting about 24FPS on Windows with a 2k x 2k image with FastImage turned on. I'd like to clean up ContrastCalculator.java a bit, since it now has eight subtly different versions of the same function, but by my tests everything is working correctly and we've reattained the speed we used to have. In fact, on OSX we're now vastly faster than we were even in 1.4.

closed 2016-02-03 by ChrisWeisiger