micro-manager/micro-manager #160: mm2: Image.getBytesPerPixel() is leaky abstraction

marktsuchida, 2016-01-30:
It returns `4` for an RGB image. But the fact that 32 bits/pixel is used to store RGB is an implementation detail. We should be able to switch to 24 bits/pixel internal representation in the future.

What would be the consequences of deprecating this method, in favor of a new `getBytesPerComponent()` (which would return `1` for an RGB image)?

ChrisWeisiger, 2016-02-01:
The information given by this method is absolutely necessary for low-level pixel manipulation, where you need to know how many bytes to step forward to get to the next pixel (the bytes-per-pixel is the "stride" between pixels). I don't think we can reasonable characterize it as a leaky abstraction since some use cases require that information.

I'm open to suggestions for better names for it though. And getBytesPerComponent() isn't unreasonable to have, though I'm not certain what it would be used for.

marktsuchida, 2016-02-03:
Maybe just add `getBytesPerComponent()` then, although let's not for now (there will inevitably be some design work to be done the next time we add support for a new pixel format, whether it is 48-bit RGB, 32-bit unsigned, float, or packed 12-bit (the last one being a joke, of course)).

I guess it is reasonable to have the current `getBytesPerPixel()` so long as it is characterized as providing information on the return value of `getRawPixels()`, instead of pretending to be a property of the image data.

closed 2016-02-03 by marktsuchida