micro-manager/micro-manager #161: mm2: RGB histogram bugs

marktsuchida, 2016-01-30:
- With the demo camera in 32bitRGB mode, StripeWidth=0.1, exposure = 100 ms, I see the overall hue change in the image, but only ever see the blue curve in the histogram (red and green invisible).

- There is what appears to be a spurious peak at zero, even though none of the image pixels are zero.

- The yellow vertical bar indicating the intensity of the pixel being moused over seems to misbehave in RGB (I don't think we need it in RGB).

Tested on Windows.

ChrisWeisiger, 2016-02-02:
I've fixed the first problem, which turned out to be an error in the histogram calculation code -- all three components were getting the same data calculated for them, causing their graphs to precisely overlap. I'm unable to reproduce the second issue. The third issue is fixed: there was a bug in calculating the pixel intensity for multi-component images.

marktsuchida, 2016-02-03:
The second issue seems to have fixed itself.

closed 2016-02-03 by marktsuchida