micro-manager/micro-manager #162: mm2: Autostretch broken for 8-bit images

marktsuchida, 2016-01-30:
With the demo camera in 8-bit mode, StripeWidth=0.1, exposure = 100 ms, Autostretch and Auto once give incorrect ranges in the histogram view. Also not sure how the images are actually being scaled in this case.

marktsuchida, 2016-01-30:
It might be that autostretch is working correctly but the intensity range is drawn incorrectly over the histogram.

marktsuchida, 2016-01-30:
It might be that autostretch is working correctly but the intensity range is drawn incorrectly over the histogram.
Stretch the Inspector horizontally to see this symptom.

ChrisWeisiger, 2016-02-01:
Display of 8-bit images seems to have problems in general. The contrast settings are correctly calculated, but the image is drawn as far too dim; effectively the display appears to be locked in a 16-bit display mode even if the data is 8-bit. Note that composite and RGB 8-bit images both work, at least in terms of producing a usable image; the histogram display is still wrong.

ChrisWeisiger, 2016-02-01:
Both of these issues should now be fixed.

closed 2016-02-01 by ChrisWeisiger