micro-manager/micro-manager #164: mm2: Histogram artifacts with narrow histogram panel

marktsuchida, 2016-02-03:
For example,
DemoCamera, RGB32, 512x512, 0.1 StripeWidth, 100 ms exposure -> multiple spikes when Inspector made as narrow as possible, spikes go away when making wide enough.

It appears that the histogram bin count & width are determined by the width of the histogram panel (or at least that would seem to explain the effect). As far as I can tell, this only happens in RGB mode, not sure why.

ChrisWeisiger, 2016-02-03:
It's probably a bug with how histograms are displayed when there's fewer horizontal pixels than there are bins in the histogram data. Will take a look.

ChrisWeisiger, 2016-02-03:
This is now fixed. The issue was actually one of how we were handling transformation of the raw histogram data for display purposes, which was able to combine bins together when we were zoomed out, but not able to handle things properly when we zoomed in.

closed 2016-02-03 by ChrisWeisiger