micro-manager/micro-manager #169: mm2: Timestamps are absolute and not always completely visible

nicost, 2016-02-04:
Using the TimeStamp overlays in 2.0, I get timestamp that look like: 2015-11-25 16:32:43 -800.  I don't think there will be many people who like that format.  Almost invariably, one will want a relative timestamp, either starting at 0:00, or at an arbitrary time point.  It may also be nice to let the use choose between ms, s, min, and hr as the time scale.  
In addition, when I choose positions on the upper or lower right, half the time stamp disappears from view (i.e, I only see 2015-11-25).

ChrisWeisiger, 2016-02-04:
We should probably have a checkbox to switch between absolute and relative time (and default to relative).

The overlay includes controls for setting the X/Y offsets, so you should be able to "fix" things so that the entire timestamp is visible. However, it shouldn't be clipping itself in the first place (i.e. an offset of 0 should ensure that the edge of the timestamp aligns with the edge of the image).

ChrisWeisiger, 2016-02-05:
I've fixed the drawing issues and added a "relative timestamp" mode that shows the elapsedTimeMs metadata field instead of the receivedTime metadata field (the original timestamp format). The latter format also now omits the timezone.

closed 2016-02-05 by ChrisWeisiger