micro-manager/micro-manager #170: Beanshell running old versions of functions

ChrisWeisiger, 2016-02-05:
Repro case: 

    void printme(int i) {
        print("I is " + i);
    }
    for (int i = 0; i < 10; ++i) {
       printme(i);
    }

Run this, then modify printme to change the print statement, and run it again; note the output still references the old function. Tested in 2.0, 1.4.22, and 1.4.13 and it all reproduces.

ChrisWeisiger, 2016-02-05:
Alternate repro doable without the scripting panel (i.e. just with the REPL):

`void run(i) {print("Hello, " + i);}`
`void run(i) {print("Goobye, " + i);}`
`run(0)`

will print `Hello, 0`.

This also reproduces in Fiji's Beanshell scripting system, so it is probably a bug in Beanshell itself.

ChrisWeisiger, 2016-02-09:
Per http://forum.imagej.net/t/alternative-to-beanshell/805 Curtis recommends we investigate the possibility of using Groovy instead.

ChrisWeisiger, 2016-04-13:
Upgrading to Beanshell 2.0b6 fixes the first repro case, but not the second one.

marktsuchida, 2016-06-30:
In my hands BeanShell 2.0b6 fixes neither repro case.

Added `mm.scripter().resetInterpreter();`, which can be used to clear the interpreter namespace. This can be used to work around the first repro case. It can be called at the beginning of a script during debugging to ensure method definitions are not stale.

Fixing the second repro case would of course require fixing BeanShell itself.

nicost, 2020-04-04:
@marktsuchida  OK to close?

marktsuchida, 2020-04-04:
Yeah, the only fix would be to move away from a buggy and unmaintained language.

closed 2020-04-04 by marktsuchida