micro-manager/micro-manager #173: Reloading HW config buggy when config group of same name exists

marktsuchida, 2016-02-08:
E.g. Load the demo config, then load a real config containing 'Channel' group. The contents of the Channel group are wrong.

ChrisWeisiger, 2016-02-09:
Whatever's going wrong here isn't specifically to do with config group names. If I create a config file with a "Channel" config group but nothing else, then switch in the demo config file, then the first config group (which happens to be "Camera") will have the old config file's Channel group options. Once I try to interact with any of the config groups, I'll get an error complaining that I tried to set the Camera group to an invalid value, after which the UI will start behaving itself properly.

marktsuchida, 2020-11-26:
@zhangxiaosa and I cannot reproduce this with current versions.

closed 2020-11-26 by marktsuchida