micro-manager/micro-manager #181: mm2: Window positions are not remembered unless the window itself is closed before exciting mm

nicost, 2016-02-10:
To reproduce: 
- Start Micro-Manager
- Select Tools > Stage control... (other windows seem to behave the same)
- Change the position of the Stage Control dialog, and remember where it is
- Without closing the Stage Control dialog, close Micro-Manager (either by clicking the close icons of the MM or ImageJ window or selecting File > Exit)
- Restart MM, open Stage Control dialog and note that it does not appear at the position it was last, but rather at the position it was first.
This is a regression from 1.4, which behaves as expected.

ChrisWeisiger, 2016-02-10:
This is not a regression from 1.4, which behaves exactly the same way, as I just tested. Please only call something a regression if you have explicitly verified that the previous version of the program behaves differently.

MMFrame has a WindowListener that listens for the windowClosing callback and records the window's position at the time that it is closed. It is in fact trivial to replace the WindowListener with a ComponentListener that listens for the componentMoved callback and records the window's position. I'm not certain why we were using windowClosing instead of componentMoved; in any case this is now fixed.

closed 2016-02-10 by ChrisWeisiger

nicost, 2016-02-10:
On 2/10/16 7:49 AM, ChrisWeisiger wrote:
>
> This is not a regression from 1.4, which behaves exactly the same way, 
> as I just tested. Please only call something a regression if you have 
> explicitly verified that the previous version of the program behaves 
> differently.
>

Actually, as I wrote, that is exactly what I did, and 1.4 correctly 
remembered the position of the window for me, so it seemed quite 
appropriate to name it a regression. Also, if terse response like this 
are going to be the norm, I am certainly not continuing to volunteer 
these bug reports.

ChrisWeisiger, 2016-02-10:
Well now, that's interesting, because it doesn't for me, running the current 1.4.22 release. And looking at the MMFrame code in 1.4, there's no way that it could possibly be remembering the positions of windows that have not had their `windowClosing` callback invoked. Note that the use of the `windowClosing` callback dates back to 2014-11-19 (see commit 4c57efd8c109276710c6d79e06faae67fb407986 ); prior to that point users of the MMFrame class had to manually call the save/load methods using their own adapters/listeners. I suppose it is possible that a version prior to that date had some MMFrames using the `componentMoved` callback (and others potentially using the windowClosing callback instead), though I see no evidence of this.

What version of 1.4 are you able to reproduce the desired behavior with?

nicost, 2016-02-10:
On 2/10/16 8:33 AM, ChrisWeisiger wrote:
> What version of 1.4 are you able to reproduce the desired behavior with? 
Although I am not sure what the purpose of this discussion is: it works 
for me with 1.4.x dev (and has worked for years in the past). Tools > 
Stage position list, move the window, close Micro-Manager by clicking 
the red icon in the top left, re-open and check that the Stage position 
list is at the expected position.

ChrisWeisiger, 2016-02-10:
Argh.

Okay, I was not reading your posts correctly and thought you were talking about the Stage Control window (formerly under plugins/Device Control). My mistake; I apologize for that. I can reproduce that in 1.4 the Stage Position List does correctly remember its location -- and for that matter it does as well in 2.0, even though it's an MMDialog, not an MMFrame, and thus is unaffected by my earlier fix to MMFrames. Indeed, testing without that fix, the Stage Position List dialog still remembers its position. There's no clear reason for _why_ it should be remembering its position; it doesn't listen for the `componentMoved` callback, just the `windowClosing` callback, same as MMFrame had been. That callback isn't being invoked, nor does the PositionListDlg do anything special to recognize when it's been moved, as far as I can tell.

So my irritation earlier is because I thought you were reporting as a new bug in 2.0 a behavior that has been with us forever, viz. that MMFrames only remembered their locations when closed. Instead we have a different behavior, that MMDialogs only remember their locations when closed, that I cannot reproduce, as for me the dialog always remembers its position in any version I've tested.

marktsuchida, 2016-02-10:
@ChrisWeisiger Note that the position is also saved in the `dispose()` method. Could that be it?

nicost, 2016-02-10:
This bug has not been with us for ever.  It worked well for years.  If it broke in 1.4, then it broke quite recently (I certainly would have noticed otherwise).  Nevertheless, this discussion is not very helpful, point is that we all want it to work.

marktsuchida, 2016-02-11:
I wonder if whether `dispose()` gets called is sensitive to something subtle that goes on while exiting. That kind of thing could be dependent on the installation.

In any case, switching to an implementation that uses `componentMoved()` might be best, as it will (1) save the position even if the program crashes and (2) simplify the logic because there is only one event to listen for (instead of `windowClosing()` event and `dispose()` method).

Furthermore, it will allow an implementation that is decoupled from the `JFrame` or `JDialog` subcalss, so that you can remove the requirement to subclass `MMFrame` or `MMDialog` (which is good in case the window in question needs to subclass some other class).

Actually we already have code for this idea in the 1.4 `org.micromanager.utils.GUIUtils.recallPosition()` static method, which is only used in a few of Arthur's plugins (that method has some issues, is confusingly named, and also lacks additional functionality that `MMFrame` has, but could be fixed and extended). Using a static function like this is a one-liner during setup of the window.