micro-manager/micro-manager #184: mm2: x, y, and z position are reported incorrectly in the image metadata

nicost, 2016-02-10:
Start MM 2.0, move the xy and z stages to a new position using the Stage Control window (check the new positions in the MM status window) and snap an image.  Find the XYZPositionUm keys in the metadata.  For me the values are always 0.0.

ChrisWeisiger, 2016-02-10:
This is now fixed. The problem was that the mechanism used for the metadata display was populating certain fields with default values if they were not available; the X/Y/Z positions are not set in the metadata, so they were being defaulted to 0 each.

They are still not set in the metadata, so this "fix" will simply cause them to not show up. Definitely an improvement over showing the incorrect value; whether or not images acquired outside of the acquisition engine should have fully-populated metadata (including not just things like the stage position, but also the scope state) is a broader question that I've filed ticket #186 for. I believe this issue can now be closed.

closed 2016-02-10 by ChrisWeisiger

nicost, 2016-02-10:
On 2/10/16 11:44 AM, ChrisWeisiger wrote:
> They are still not set in the metadata, so this "fix" will simply
> cause them to not show up. Definitely an improvement over showing the
> incorrect value; whether or not images acquired outside of [...]
Nice.  It will surely be super handy to have a single method to add all 
metadata known to the system to the image metadata, rather than having 
to hand code that every time again.  In fact, my question is: how does 
one now go about fully populating the metadata including stage positions 
outside of the acquisition engine?

ChrisWeisiger, 2016-02-10:
Currently your only option is to hand-populate each field. E.g. create a `MetadataBuilder` and then call `builder.xPositionUm(core.getXYStagePosition().x)`. Assuming we decide to implement fully-populated images as per #186, there would be API methods to produce fully-populated images.