micro-manager/micro-manager #185: Add "Gather Windows" function

ChrisWeisiger, 2016-02-10:
In the event that a window is somehow placed off of any display, there should be a keyboard shortcut and/or menu option (ideally something that's always reachable) that moves all windows onto the current primary display.

ChrisWeisiger, 2016-02-17:
We should leave this to the operating system, e.g. http://superuser.com/questions/216277/windows7-how-to-gather-windows-from-secondary-monitor indicates that on Windows, an offscreen window should be able to be moved via right-clicking the taskbar icon and using the arrow keys. OSX presumably has similar tools.

closed 2016-02-17 by ChrisWeisiger

marktsuchida, 2016-02-19:
Let's hope this is sufficient. For Windows, the right-click on task bar entry method works for most windows, and one would hope that windows that do not show up in the task bar (e.g. the splash screen) will always be positioned reasonably.

For OS X I'm not sure there is a good method that works with Java apps. Normally either (1) switching between mirrored vs non-mirrored or (2) plugging or unplugging the Mini-DisplayPort/Thunderbolt connector should effectively gather the windows, but this issues was created in response to a case where that didn't happen....