micro-manager/micro-manager #186: Should we populate all images with complete metadata?

ChrisWeisiger, 2016-02-10:
Currently, we have three primary ways to generate images:

1) Snap a TaggedImage using MMCore.snapImage(), then grab it using MMCore.getTaggedImage().
2) Grab an TaggedImage from an ongoing sequence acquisition using MMCore.getNBeforeLastTaggedImage()
3) Receive a TaggedImage from the acquisition engine during an MDA.

In all three cases, we then convert the TaggedImage to a DefaultImage and its corresponding DefaultMetadata; the DefaultMetadata parses fields out of the TaggedImage's JSON. Each of the three approaches populates different metadata fields. Options 1 and 2 are only marginally different (chiefly, option 2 includes an "image number" field), but option 3 includes vastly more information, including basic fields like the X/Y/Z stage position and also a more detailed breakdown of the exact state of the microscope at the time the image was acquired (well, the exact state as subject to the engine's caching rules).

Should we be providing such detailed information for all images that pass through the Java layer? Eventually we will have a new acquisition engine, at which point presumably this will happen automatically, but in the meantime we could consider manually patching options 1 and 2 above to fill in more detailed metadata. The only place in the Java layer that generates new Image objects based on TaggedImages produced by the Core is the SnapLiveManager -- though of course any third-party code that speaks to the Core directly instead of by using our API will not benefit from such changes.

nicost, 2016-02-10:
Yes please!  It could be nice to not do that automatically, but have a function (possibly a variant of the convertTaggedImage api call) that does this for the caller.  That way, one can opt-in to get the metadata, but make it very easy to get them.  In fact, without such a method, every script/plugin writer will have to write the same (long and boring) code again, leading to duplication and possible errors. So, I vote for doing this!

marktsuchida, 2016-02-10:
I agree.

ChrisWeisiger, 2016-02-12:
I've now added `AcquisitionManager.generateSummaryMetadata()` and `AcquisitionManager.generateMetadata()` methods that fill in all information possible, excluding the exhaustive list of device property values that the acquisition engine provides. The changes associated with these commits also did a fairly substantial amount of refactoring (in particular, Nico, the ASIdiSPIM plugin has been modified again), but as far as I can tell everything still works.

ChrisWeisiger, 2016-02-16:
Do we consider it necessary to have a method that populates in the complete state of the microscope -- that exhaustive list of device property values?

nicost, 2016-02-16:
I think that it is essential that all data saved in Micro-Manager has a copy of that exhaustive list.  Preferably, each image only stores the metadata that were different from the ones in that list when the image was taken.  It should always be possible to reconstruct the complete state of the system from the metadata.

ChrisWeisiger, 2016-02-16:
I agree with you entirely, however my question was whether the API should expose a method that generates a Metadata object that includes that information, which is a different question. All images generated by the acquisition engine get that metadata "for free" (aside from the costs of building up the Metadata object), but such is not currently easily done via the API.

As for differential-based methods of generating Metadata objects, that's a design problem that I think is outside scope for this issue. In principle, the Builder pattern we use for generating image metadatas during acquisitions will take care of that for us, if not for third-party code.

nicost, 2016-02-16:
So, how are scripts/plugins, etc.. supposed to add that list of metadata to the acquisitions that they manage?

ChrisWeisiger, 2016-02-16:
Okay, then, I assume your answer to my question is "yes" and we've just been talking past each other.

ChrisWeisiger, 2016-02-17:
`AcquisitionManager` a) has been moved to `org.micromanager.acquisition`, and b) has had its `generateMetadata` method modified to take a boolean which, if set to true, will populate the image with the current state of the system state cache.

We may want to consider changing how we structure this; we could equivalently have a method that produces a `PropertyMap` containing the contents of the system state cache, and which users could manually insert into a `Metadata`'s `scopeData` field, or use for other purposes as they see fit.

ChrisWeisiger, 2016-02-19:
After discussion with Mark, we'll leave the API as-is for now. In future we can add a more direct "generate a PropertyMap from a core Configuration instance" method if we decide it's necessary.

closed 2016-02-19 by ChrisWeisiger