micro-manager/micro-manager #189: mm2: Multi-channel, multi-position images are not displayed correctly

nicost, 2016-02-10:
Steps to reproduce:
- Use MM 2.0 with the demo configuration (set the camera to display noise to make image changes more obvious)
- Run an acquisition with 2 channels and multiple positions
- Make sure that Composite Color mode is selected
- Move the position slider and observe that the displayed image does not change even though the status line and metadata change as expected

When switching to Grayscale or Color mode there is a change in the displayed image, demonstrating that the problem is in the viewer, not in the acquisition itself.

ChrisWeisiger, 2016-02-12:
Thanks for the report! That's kind of important to have working, and I'm surprised nobody noticed it before now. It's fixed now.

closed 2016-02-12 by ChrisWeisiger

nicost, 2016-02-12:
Confirmed.  Thanks!