micro-manager/micro-manager #190: mm2: per image metadata in Inspector Window still shows more than it should

nicost, 2016-02-11:
The per-image metadata display in the Inspector Window now seems to have a global memory rather than a per-image memory.  This effect can be demonstrated by the following steps:
- Start mm with demo configuration
- Run a MDA with multiple timepoints, 0 ms interval
- There are 5 properties shown in the per-image Metadata are, all appropriate (they all change)
- Now run a MDA with no time points but 2 channels, 8 properties are shown (one of them is Camera, which seems erroneous)
- Now switch back to the first data set and see that 9 properties are shown, 4 of which should not be there.  These are all related to the channels that changed in the second acquisition

Closing the inspector window and opening a new one results in all metadata in the image being shown (i.e. a list of ~100 properties), so that is not work around either (and likely a related bug)

ChrisWeisiger, 2016-02-12:
Thanks for the report; this is now fixed.

closed 2016-02-12 by ChrisWeisiger

nicost, 2016-02-12:
Almost perfect!  Only issue remaining is that when switching windows, the user has to force an image redraw to update the metadata that are being displayed.  Try first snapping a single image using the multi-camera configuration (very few metadata), then a multi-channel z-stack (many metadata).  When switching back to the single snapped image I first see the long list of metadata and have to trigger an update by changing the channel slider (only possible with the multi-camera) to get the correct number of metadata.