micro-manager/micro-manager #192: mm2: AutofocusManager is an internal structure, yet exposed in the api

nicost, 2016-02-11:
AutofocusManager is currently in org.micromanager.internal.utils, yet it is accessible through the mm.getAutofocusManager() function and itself refers to autofocus devices in org.micromanager.AutofocusPlugin, so it makes a lot of sense to move it to an "external" package.  A problem is that it throws MMException (which is in the same internal package).  I do think that there is a more general need for an "external" MMException class.

ChrisWeisiger, 2016-02-11:
The general trend in 2.0 is to avoid using MMException wherever possible -- it's too generic to provide useful information about what went wrong unless you examine the string text of the exception (which you can't realistically do at runtime). Instead, we want to use more specific exceptions that describe what went wrong, e.g. DatastoreFrozenException, IllegalArgumentException, etc.

I agree that the autofocus API is not fully-converted to 2.0 though, and that includes the numerous references to MMException and MMScriptException (though in that respect it's not alone, e.g. AcquisitionManager has the same problem).

nicost, 2016-02-11:
OK, but what about AutofocusManager being internal?  That is the first and main problem.  Any solution to the Exception problem is fine (and I do not mind general exceptions with text messages only, as long as they can be caught and shown to the user).

marktsuchida, 2016-02-11:
Since AutofocusPlugin is already in the 2.0 API, it looks like all that is needed in the API is the small number of methods in AutofocusManager (though probably not in exactly the same form, certainly not a class).

I would consider the show/hide of AF options dialog to belong to the app GUI API, not the autofocus API (which should work headless).

@ChrisWeisiger Do you think that's feasible?

ChrisWeisiger, 2016-02-11:
Yes, what we should probably do is add an AutofocusManager interface to org.micromanager, and then have the org.micromanager.internal.utils.AutofocusManager class implement that interface (and probably rename the implementation to DefaultAutofocusManager).

ChrisWeisiger, 2016-02-12:
This is now done; see https://github.com/micro-manager/micro-manager/blob/mm2/mmstudio/src/main/java/org/micromanager/AutofocusManager.java for the new API interface.

Nico: the ASIdiSPIM plugin has been tweaked slightly since the methodology for setting the current autofocus device is slightly different now. Sorry for the inconvenience. Also, while I tested each of the software autofocus devices and they appear to work (with demo hardware), if you have access to a hardware autofocus device and the time to test, I'd appreciate making certain that it still behaves properly. Thanks!

closed 2016-02-12 by ChrisWeisiger