micro-manager/micro-manager #193: mm2: Allow vertical resizing of histograms

marktsuchida, 2016-02-12:
Strong request from certain quarters :)

This could be done one of two ways:
- Add to the gear menu in the inspector an option to select one of several vertical size
- Create a separate "big histogram" plugin, with its own resizable window, to display just the histograms

nicost, 2016-02-14:
The same problem applies to the summary and per image metadata tables.  Not being able to resize them makes them very difficult to work with.

ChrisWeisiger, 2016-02-18:
The histogram, metadata, and comments panels in the Inspector can now grow to fill available vertical space. This took over two days to get working properly.

nicost, 2016-02-19:
The histograms behave beautifully!  Very nice, and that must have been a pain to get right.  Metadata panel also looks a lot better, but I did notice a few glitches (happy to open new Issues if you prefer):
* There is a red-blue dashed outline along the metadata panel.  Is that on purpose or left over debug code?
* When scrolling slowly, some text in the tables gets garbled and unreadable (only in Java 1.6, in 1.8 it looks great!)
* When checking "Show unchanged properties" the long list is shown as expected, but then unchecking it does not reduce the list (i.e. all the unchanged properties are still shown).

This is great progress!

ChrisWeisiger, 2016-02-19:
Feel free to re-open issues if they aren't completely resolved.

The dashed outline is gone, and the display bug with the checkbox should be fixed. I was unable to replicate the garbled text in Java 1.6, on OSX or Windows. How slowly were you scrolling?

marktsuchida, 2016-02-20:
The reason I didn't initially suggest that the histograms simply grow according to the vertical size of the inspector is that it gets really complicated when you have multiple panels open in the inspector.

For example, how do I get the desired vertical size for the histogram(s) and metadata when I have both of those panels open?

Under the current way things are, perhaps we need draggable dividers between sections, or some equivalent thereof.

ChrisWeisiger, 2016-02-20:
I feel like implementing a manually-resizable system would create a lot of work for negligible gain. If we need finer-grained controls than we have now, specifically for the histograms, then the solution should be to have an optional dedicated histogram window, as we discussed earlier.

Keep in mind that getting the resizing we have now took over two days worth of work. I'm not exactly thrilled at the prospect of diving back in again.

nicost, 2016-02-21:
I quite like the way things work right now.  In practice, I usually focus on either the histograms or the metadata, and never feel the need to see them at the same time, and the current resizing behavior is fine for that workflow.

marktsuchida, 2016-02-21:
I think we need an overall design goal here. If the goal is to provide maximum flexibility for resizing each panel but not really providing support for interacting with multiple panels at the same time (and I think that's not an unreasonable goal, just not what I originally had in mind given how the Inspector worked until recently), then we should "auto-hide" all other panels when the user opens one panel. That would save the user from having to close the previous panel before (or after) opening another one, and would also make it easier for the program to retain the user's preferred height for each panel (which I think is essential, at least for the histograms). Auto-hiding all but one vertical element is also a UI behavior that is fairly common in (native and web) apps, so it won't be surprising to users.

I think the main UI issues that result in frustration are arbitrary limitations (lack of control over an obvious parameter), inconsistency, and unpredictability. The original Inspector had an arbitrary limitation (inability to change the vertical size of any of the panels) but was logically consistent and very predictable (I do not propose going back to it).

I have two problems with the current behavior:

1. It still has arbitrary limitations in practice, since as soon as you have more than one panel open, there is no ability to resize any individual panel.
2. The vertical resizing behavior is too complex for the user to predict, and there are no simple rules a user can follow to preserve their favorite height for any given panel. For example, if I like to have my histograms be any height other than the minimum allowed, there is no way to access the metadata without worrying that I will lose that histogram height setting (and in fact I do lose the height). In a different example, if I have only the metadata open and have it vertically stretched from top to bottom of my screen, then open the histograms, I see not only the metadata panel but the entire Inspector shrinking vertically. This is very frustrating.

So, if it's looking like a completely individually resizable arrangement is impractical, my vote at this point would be to introduce the simplifying constraint that only one panel can be open at a time -- but instead make sure the height of each panel (height per channel in the case of the histograms) is remembered by the application. This does introduce the (hopefully not so severe) limitation that the user cannot view two panels simultaneously (except by opening a second inspector), but all behavior should be predictable after a few tries by the user, and there will be no frustration arising from the histograms or metadata panels resizing when the user doesn't want them to. A minor added benefit might be an overall less crowded screen.

Well, at least that's how far my thinking has gone so far. It's possible I'm underestimating the harm caused by not being able to view multiple panels at the same time, so further discussion is welcome....

ChrisWeisiger, 2016-02-22:
My inclination at this point is to say that we've reached the limit of what we can accurately define as "desired behavior" without feedback from a broader base of users. I agree that not remembering the histogram sizes is probably not desirable, but the tradeoffs involved in any solution you or I can think of mean that I'd really want to get some more detailed user feedback before committing to any one approach.

There is of course also the potentially substantial technical challenge of actually implementing whatever we decide should be done. At some point we start hitting diminishing returns on what features are worth implementing. Given the effort involved in implementing the most recent pass (despite how simple it appeared to be conceptually), I'm loathe to try for further tweaks unless we're really certain they're what we want.

ChrisWeisiger, 2016-02-26:
I've made some tweaks to the Inspector so that it's better about trying to use the size that the user last set for a given panel. Feedback is welcome.

ChrisWeisiger, 2016-03-24:
Okay, I think we can put this one to bed.

closed 2016-03-24 by ChrisWeisiger