micro-manager/micro-manager #199: mm2: Image viewer status line disappears from the screen under certain conditions

nicost, 2016-02-14:
When the Image Viewer contains many channels, the space reserved for displaying the pixel intensities pushes the status line (channel, z, time point, etc.) information to the right, eventually beyond the area being displayed.  This can be most easily seen by running an MDA with many channels (i tried 4 channels with 2 cameras > 8 total).  Options are to display either the status line or the pixel intensities (depending on where the cursor is) or to push the status line to the left when the cursor is not inside the image.

ChrisWeisiger, 2016-02-18:
A third option would be for the user to simply make the window wider if they need to see that information. Consider that we are potentially "vulnerable" to this issue no matter what we do, as the user may decide that they want to collect 64x64 images, or to view their images while zoomed way out. If the window is too small to contain the information in the status line, then it will get truncated, and that's simply a fact of life.

Do we have reason to believe that users are collecting data with >4 channels in practice? In my testing, with a 512x512 display window at 100% zoom, a 4-channel multi-T/Z/P acquisition is able to show everything except for the camera name in the status display without needing to make the window wider than the image data. That seems pretty good to me.

My concern with showing/hiding the pixel info display (and moving the status text over to compensate) is that it means that the position of components will depend on where the mouse is, which may make the display window harder to visually "parse". It _will_ also make the layout code slightly more complicated, and that's already a big problem with the display window. In short, I'm not convinced that the gain is worth the cost.

nicost, 2016-02-18:
As in my example, it is actually quite easy to get > 4 channels (especially using multiple cameras), and this will happen often enough in practice (I ran into it using the diSPIM).  In practice, it is undoable to size the window such that the status line is shown (let alone the user who is not even aware that she is supposed to be doing that to get to see that information, plus the ugliness of a a window that needs to much wider than the image to show the status line).

If you do not like the solution of moving the status line to the left when the pixel intensities are not shown, there are other solutions:
* Only show the pixel intensity of the active channel.  I have never had the desire to compare the pixel intensities of multiple channels simultaneously, and also find it hard to mentally parse a line with many numbers, so only showing the intensity of the active channel (using the channel slider) may be nicer in any case
* Place the status line and pixel intensities on separate lines (at the cost of slightly higher display windows)
* Place the status line to the left of the pixel intensities.  After all, if one can read pixel intensities but not what at which channel or timepoint that information came from, then the pixel intensity is no very useful information.  

I think that my vote would be for status line to the left (so that it is always visible), pixel intensity only of the active channel to the right thereof.

ChrisWeisiger, 2016-02-25:
After discussion with Mark, we've rearranged the layout of the various info text lines in the display window. In particular, the pixel info display has been moved to the top-right corner, which should free up plenty of space for the image info text.

closed 2016-02-25 by ChrisWeisiger