micro-manager/micro-manager #202: mm2: AutofocusBase is in an internal package but only used by external packages

nicost, 2016-02-18:
By its very nature, AutofocusBase is only useful to "external" code.  One option is to move it to an autofocus package when restructuring the organization of the autofocus code.

marktsuchida, 2016-02-20:
Also, according to Java naming conventions, it should probably be renamed to `AbstractAutofocusPlugin`.

ChrisWeisiger, 2016-04-08:
`AutofocusPlugin` and `AutofocusBase` both need to have some substantial work done to them to bring them up to the standards of the rest of the API. I don't really have familiarity with them, though, so would appreciate some insight if available on the following questions:

* Why do so many methods in `AutofocusPlugin` throw `MMException`? None of the exceptions are documented. For preference, we should not throw such "generic" exceptions, instead preferring more specific ones like `IllegalArgumentException` and making new exception classes when necessary (c.f. `DatastoreFrozenException`). Even throwing `Exception` is preferable IMO because then at least users don't need to import the `MMException` module (which is in internal.utils).

* It looks like `getNumberOfImages`, `getVerboseStatus`, and `getCurrentFocusScore` are not used anywhere. Can we safely remove them?

* `enableContinuousFocus`, `isContinuousFocusEnabled`, and `isContinuousFocusLocked` are only used by hardware-based autofocus devices (i.e. these methods are not called for software autofocus systems). This seems to make sense to me; can we separate them out of `AutofocusPlugin`? Ideally the Java plugin should be strictly for software-based autofocus systems; the fact that we also use it behind the scenes for hardware autofocus devices should not be exposed here.

* Is there some guideline for the number returned by `computeScore`? Would it make sense to pass an `Image` object to the method rather than an `ImageProcessor`? `ImageProcessor`s can easily be produced by the `Studio`, which all plugins have access to automatically.

* The property system relies on `PropertyItem`, which is also an internal module and which would be very confusing if exposed in the API due to the existing `PropertyMap` which has nothing to do with it. Opinions?