micro-manager/micro-manager #205: mm2: "Acquisition started, waiting for images" box stays around if MDA fails

marktsuchida, 2016-02-19:


ChrisWeisiger, 2016-02-19:
Can confirm; a simple way to replicate this on demo hardware is to paste this into the script panel: `mm.acquisitions().attachRunnable(0, 0, 0, 0, new Runnable() {public void run() {throw new RuntimeException("Oops");}});` . I believe the problem is that the acquisition halts before the StatusDisplay (the "waiting for images" popup) has a chance to register for events, which means it never gets the AcquisitionEndedEvent and thus never realizes it needs to disappear.

One potential fix is for the StatusDisplay to periodically poll the AcquisitionManager to determine if an acquisition is still running. That seems hackish but would do the job. Another would be to try to rework how we go about setting up an acquisition so we can set up all of the framework around running an acquisition before we actually start it. The big sticking points appear to be a) that we only get access to the `BlockingQueue<TaggedImage>` that the `DefaultTaggedImageSink` needs by invoking `run()` on the AcquisitionEngine, and b) that the AcquisitionEngine is the source for our SummaryMetadata, and I think that `run()` needs to have been invoked for the SummaryMetadata it returns to be valid.

An approach whereby we call some kind of `setup` method on the AcquisitionEngine, then hand it a BlockingQueue that we create ourselves, and _then_ invoke `run()`, would allow us to set up all of our Java-side entities prior to the acquisition actually getting started, and thus would be more resilient against immediate failures. But that would require some degree of reworking how we start up acquisitions, and I'm not about to try to tackle that on a Friday afternoon.

marktsuchida, 2016-02-20:
@ChrisWeisiger The latter approach might be good in the long term, but for this specific problem would it be sufficient for the StatusDisplay to check once (no need for polling), right _after_ registering for events, whether the acquisition is still running?

ChrisWeisiger, 2016-02-20:
Good call; will do that on Monday.

ChrisWeisiger, 2016-02-22:
The latter fix is now implemented. I'll save re-architecting our acquisition startup for when we have a new acquisition engine to do it with, since doubtless that entire system will be overhauled anyway.

closed 2016-02-22 by ChrisWeisiger